beagleboard/linux #157: Audio cape not working on Debian 9.3(4.9 kernel)

jjghjjgh, 2018-03-27:
Hello,
I am able to load  BB-BONE-AUDI-02 on 8.7 Debian(4.4 kernel) as shown below.
8.7 debian IoT( 4.4.54-ti-r93 kernel)::

aplay -l ---> no devices found

when i changed in uEnv.txt(cape_enable=bone_capemgr.enable_partno=BB-BONE-AUDI-02),then
root@beaglebone:~# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: B [AudioCape Rev B], device 0: davinci-mcasp.0-tlv320aic3x-hifi tlv320aic3x-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
root@beaglebone:~# dmesg | grep cape
[    0.000000] Kernel command line: console=ttyO0,115200n8 bone_capemgr.enable_partno=BB-BONE-AUDI-02 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait
[    3.640488] bone_capemgr bone_capemgr: Baseboard: 'A335BNLT,00C0,4717BBBK0E43'
[    3.647847] bone_capemgr bone_capemgr: compatible-baseboard=ti,beaglebone-black - #slots=4
[    3.694039] bone_capemgr bone_capemgr: slot #0: No cape found
[    3.738032] bone_capemgr bone_capemgr: slot #1: No cape found
[    3.782031] bone_capemgr bone_capemgr: slot #2: No cape found
[    3.826031] bone_capemgr bone_capemgr: slot #3: No cape found
[    3.831847] bone_capemgr bone_capemgr: enabled_partno PARTNO 'BB-BONE-AUDI-02' VER 'N/A' PR '0'
[    3.840606] bone_capemgr bone_capemgr: slot #4: override
[    3.845962] bone_capemgr bone_capemgr: Using override eeprom data at slot 4
[    3.852977] bone_capemgr bone_capemgr: slot #4: 'Override Board Name,00A0,Override Manuf,BB-BONE-AUDI-02'
[    3.862957] bone_capemgr bone_capemgr: initialized OK.
[    3.870580] bone_capemgr bone_capemgr: slot #4: dtbo 'BB-BONE-AUDI-02-00A0.dtbo' loaded; overlay id #0
the logs show that BB-BONE-AUDI-02 overlay was loaded.
But when i tried enabling same audio cape on debian 9.3(4.9 kernel) using uboot-overlay method,its hanging while booting itself,not completing the booting.

9.3 debian IoT(4.9.82 kernel)::

my uEnv.txt:
root@beaglebone:/sys/devices/platform/ocp# cat /boot/uEnv.txt
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0
uname_r=4.9.82-ti-r102
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
uboot_overlay_addr0=/lib/firmware/BB-UART1-00A0.dtbo
uboot_overlay_addr1=/lib/firmware/BB-BONE-AUDI-02-00A0.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
#disable_uboot_overlay_video=1
disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_uio (4.4.x-ti, 4.14.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks... (60000 = 384KB)
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###
#cmdline=coherent_pool=1M net.ifnames=0 quiet
#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 quiet video=HDMI-A-1:1024x768@60e
##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh

root@beaglebone:~# aplay -l
aplay: device_list:270: no soundcards found...

i tried adding BB-UART1 as well using uboot-overlay,so boot logs tell uart1 got loaded properly.
uboot_overlays: loading /lib/firmware/BB-UART1-00A0.dtbo ...
1075 bytes read in 222 ms (3.9 KiB/s)
 
But when i tried the same with audio cape,the booting hangs in between as shown below in the boot logs:
ERROR: Did not find a cmdline Flattened Device Tree
Could not find a valid device tree
** Invalid partition 2 **
** Invalid partition 3 **
** Invalid partition 4 **
** Invalid partition 5 **
** Invalid partition 6 **
** Invalid partition 7 **
switch to partitions #0, OK
mmc1(part 0) is current device
Scanning mmc 1:1...
gpio: pin 56 (gpio 56) value is 0
gpio: pin 55 (gpio 55) value is 0
gpio: pin 54 (gpio 54) value is 0
gpio: pin 53 (gpio 53) value is 1
switch to partitions #0, OK
mmc1(part 0) is current device
uboot_overlays: loading /lib/firmware/BB-BONE-AUDI-02-00A0.dtbo ...
2859 bytes read in 306 ms (8.8 KiB/s)
failed on fdt_overlay_apply(): FDT_ERR_NOTFOUND
uboot_overlays: loading /lib/firmware/BB-BONE-eMMC1-01-00A0.dtbo ...
1440 bytes read in 373 ms (2.9 KiB/s)
libfdt fdt_check_header(): FDT_ERR_BADMAGIC
failed on fdt_overlay_apply(): FDT_ERR_BADMAGIC
base fdt does did not have a /__symbols__ node
make sure you've compiled with -@
uboot_overlays: cape universal disabled, external cape enabled or detected...
loading /boot/initrd.img-4.1.15-ti-rt-r43 ...
4096241 bytes read in 274 ms (14.3 MiB/s)
debug: [console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=UUID=76ea7739-c46a-42c5-89e1-5dbb28d3a184 ro rootfstype=ext4 rootwait coherent_pool=1M quiet cape_universal=enable] ...
debug: [bootz 0x82000000 0x88080000:3e80f1 88000000] ...
ERROR: Did not find a cmdline Flattened Device Tree
Could not find a valid device tree
debug: [console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M quiet cape_universal=enable] ...
debug: [bootz 0x82000000 0x88080000:3e80f1 88000000] ...
ERROR: Did not find a cmdline Flattened Device Tree
Could not find a valid device tree
** Invalid partition 2 **
** Invalid partition 3 **
** Invalid partition 4 **
** Invalid partition 5 **
** Invalid partition 6 **
** Invalid partition 7 **
starting USB...
USB0:   Port not available.
link up on port 0, speed 100, full duplex
BOOTP broadcast 1
DHCP client bound to address 192.168.1.243 (4 ms)
Using cpsw device
TFTP from server 192.168.1.1; our IP address is 192.168.1.243
Filename 'vmlinuz-4.1.15-ti-rt-r43'.
Load address: 0x82000000
Loading: T T T T T T T T T T T T T T T T T T T T
Retry count exceeded; starting again
missing environment variable: pxeuuid
Retrieving file: pxelinux.cfg/01-98-5d-ad-ec-34-c4
link up on port 0, speed 100, full duplex
Using cpsw device
TFTP from server 192.168.1.1; our IP address is 192.168.1.243
Filename 'pxelinux.cfg/01-98-5d-ad-ec-34-c4'.
Load address: 0x80100000
Loading: T T T T T T T T T T T T T T T T T T T T

I have been stuck with this since a week,please help where am i going wrong.I couldnt find any support for loading audio cape on latest kernel. waiting for your response.
Thanks,
Jithu.

AndyW999, 2018-05-02:
Hi jjghjjgh

Yes, I am having the same problem so I did what was suggested on https://github.com/jan1s/hw-bbb-audiocape - 
Download and unzip the device tree source:
wget http://elinux.org/images/1/10/BB-BONE-AUDI-02-00A0.zip
unzip BB-BONE-AUDI-02-00A0.zip
Compile the device tree file and move it to /lib/firmware:
dtc -O dtb -o BB-BONE-AUDI-02-00A0.dtbo -b 0 -@ BB-BONE-AUDI-02-00A0.dts
mv BB-BONE-AUDI-02-00A0.dtbo /lib/firmware

Alsamixer finds a DA830 EVM and when doing speaker-test -t wav -c 2 I get I2C communication and a 25MHz clock from GPIO3_21 but no other pin changes using my logic analyser.

Can you try this as I don't have an spare BBB to try in case the SPI port is broken as there is no bit clock.

Thanks

Andy.

jjghjjgh, 2018-06-08:
Hi Andy,
Ya even i followed the same link ,compiled the dts file and copied .dtbo file to /lib/firmware but my problem is i am able to do it completely on 8.7 debian but not on 9.3 debian. In my post i have clearly mentioned everything along with the boot logs.which debian version are you using?
thanks,
Jithu

AndyW999, 2018-06-08:
Hi Jithu

I have actually given up with it and gone for an Olimex Allwinner A20 board that actually works out of the box!

I know somebody who is doing a lot of development on BBB and he said to just use a USB device as it looks like the Audio Cape is just too much trouble.

Thanks

Andy.

jjghjjgh, 2018-06-09:
Hi Andy,
thanks for your quick response.Unfortunately i cant give up BBB as my whole project is based on that.What kind of USB device are you talking about? Sorry i dint understand that part. I badly need a solution for the audio part.
Thanks ,
Jithu

AndyW999, 2018-06-09:
Hi Jithu

Make sure the audio device and HDMI audio is turned off then use something like this -

https://www.amazon.co.uk/s/ref=nb_sb_ss_c_1_11?url=search-alias%3Daps&field-keywords=usb+audio+adaptor&sprefix=usb+audio+a%2Caps%2C241&crid=30LKIIECUDVCN&rh=i%3Aaps%2Ck%3Ausb+audio+adaptor

Most USB audio dongles are supported under Linux.

Andy.

jjghjjgh, 2018-06-09:
> search-alias%3Daps&field-keywords=usb+audio+adaptor&
> sprefix=usb+audio+a%2Caps%2C241&crid=30LKIIECUDVCN&rh=i%
> 3Aaps%2Ck%3Ausb+audio+adaptor

Oh thanks.that sounds cool. I will definitely give it a try and get back to
you.
Regards,
Jithu

deeplyembeddedWP, 2019-10-29:
Has this been resolved? Anyone got the audio cape working?