beagleboard/linux #159: kernel 4.14 doesn't boot due to CONFIG_NLS_CODEPAGE_*=m

dlech, 2018-04-13:
I'm not sure what it is exactly about my configuration (maybe because I am mounting a FAT partition using `/etc/fstab`?), but 4.14.34-ti-r43 is not booting for me due to some error messages similar to this:

    FAT: codepage cp437 not found
    ...
    Cannot open access to console, the root account is locked.
    See sulogin(8) man page for more details.

    Press Enter to continue.

I fixed it with this:

```diff
diff --git a/arch/arm/configs/bb.org_defconfig b/arch/arm/configs/bb.org_defconfig
index e496c71fc5b8..aa2d4a8a004b 100644
--- a/arch/arm/configs/bb.org_defconfig
+++ b/arch/arm/configs/bb.org_defconfig
@@ -2409,7 +2409,7 @@ CONFIG_9P_FSCACHE=y
 CONFIG_9P_FS_POSIX_ACL=y
 CONFIG_9P_FS_SECURITY=y
 CONFIG_NLS_DEFAULT="utf8"
-CONFIG_NLS_CODEPAGE_437=m
+CONFIG_NLS_CODEPAGE_437=y
 CONFIG_NLS_CODEPAGE_737=m
 CONFIG_NLS_CODEPAGE_775=m
 CONFIG_NLS_CODEPAGE_850=m
@@ -2432,8 +2432,8 @@ CONFIG_NLS_CODEPAGE_874=m
 CONFIG_NLS_ISO8859_8=m
 CONFIG_NLS_CODEPAGE_1250=m
 CONFIG_NLS_CODEPAGE_1251=m
-CONFIG_NLS_ASCII=m
-CONFIG_NLS_ISO8859_1=m
+CONFIG_NLS_ASCII=y
+CONFIG_NLS_ISO8859_1=y
 CONFIG_NLS_ISO8859_2=m
 CONFIG_NLS_ISO8859_3=m
 CONFIG_NLS_ISO8859_4=m
```

I'm not sure why this is affecting me now. I was using the 4.9.x kernel from this repo with the exact same rootfs before without any problem.

I'm carrying my own patches anyway, so I don't particularly care if this gets picked up or not. I just wanted to throw it out there in case anyone else runs into the same problem or maybe knows another way to solve the issue.

dlech, 2018-04-13:
well, I just realized that *no* modules are loading at all, so that is probably the problem :confused:

RobertCNelson, 2018-04-13:
@dlech that's from:

```
CONFIG_MODULE_COMPRESS=y
# CONFIG_MODULE_COMPRESS_GZIP is not set
CONFIG_MODULE_COMPRESS_XZ=y
```
You version of kmod wasn't patched/built for compressd modules.  Here's the package patch:

https://anonscm.debian.org/cgit/users/md/kmod.git/commit/?id=083f4b0a859e462162cdee938d5dcf5274a59bfc

I back ported this to Stretch/Jessie, it cuts the size of /lib/modules/`uname r`/ by around half..

Regards,

dlech, 2018-04-13:
Thanks for the info, Robert.

dlech, 2018-04-13:
FWIW, I've just disabled the compression for now. Might have a look at the kmod patch later.

```diff
diff --git a/arch/arm/configs/bb.org_defconfig b/arch/arm/configs/bb.org_defconfig
index e496c71fc5b8..4723dc0e2ba4 100644
--- a/arch/arm/configs/bb.org_defconfig
+++ b/arch/arm/configs/bb.org_defconfig
@@ -46,8 +46,6 @@ CONFIG_MODULE_FORCE_LOAD=y
 CONFIG_MODULE_UNLOAD=y
 CONFIG_MODULE_FORCE_UNLOAD=y
 CONFIG_MODVERSIONS=y
-CONFIG_MODULE_COMPRESS=y
-CONFIG_MODULE_COMPRESS_XZ=y
 CONFIG_BLK_DEV_THROTTLING=y
 CONFIG_BLK_WBT=y
 CONFIG_PARTITION_ADVANCED=y
```

closed 2018-04-13 by dlech