beagleboard/linux #167: bone-rt-kernel --lts-4_9      missing cpu scaling_governor

patrickpoirier51, 2018-05-12:
Following the 4_14  cpu load issue https://github.com/beagleboard/linux/issues/166

I reverted back to 4_9 and loaded sudo apt install cpufrequtils
When doing cpufreq-info I get this:
cpufrequtils 008: cpufreq-info (C) Dominik Brodowski 2004-2009
Report errors and bugs to cpufreq@vger.kernel.org, please.
analyzing CPU 0:
  no or unknown cpufreq driver is active on this CPU
  maximum transition latency: 0.00 ms.

ls /sys/devices/system/cpu/cpu0/
crash_notes  crash_notes_size  of_node  power  subsystem  topology  uevent
There is no scaling_governor

RobertCNelson, 2018-05-14:
This should be fixed. (kernel dtb bug)

```
sudo /opt/scripts/tools/version.sh
```

jadonk, 2018-05-23:
did you do 
```sh
cd /opt/scripts
git pull
sudo tools/version.sh
```
first?

patrickpoirier51, 2018-05-23:
Jason , if you read the above, it wrote t; the file is there and it gets executed, there is just No Result

closed 2019-04-09 by patrickpoirier51