beagleboard/linux #169: omap2_mcspi_transfer_one off by mSec - kernel 4.4.68

vinifr, 2018-05-17:
Hi,  I have a Beaglebone Black Rev C.

I am doing a high speed ADC capture with a chip which has SPI interface. The SPI frequency is 2MHz. In a 10 seconds ADC capture writing/reading 4 bytes in SPI at 200uSec interval, the function omap2_mcspi_transfer_one is not called by kernel 200-250 times in 200uS interval. 

For some reason the kernel is not calling omap2_mcspi_transfer_one during 1.72mSec and sometimes 7mSec. What could cause this suspension of omap2_mcspi_transfer_one?

The first signal in image bellow is the select pin (CS).
SPI select pin(CS) off by 7mS: [https://drive.google.com/open?id=1gQgkuNoldyI3Xfkc0JyYTizi9Mog0Xok](url)

Bellow a example of SPI working and tranfering 4 bytes in 25uS. The first signal is CS pin.
[https://drive.google.com/open?id=1c1Lqz3Dmlx6qShi8BnTiUQlX_QGsvlNT](url)

RobertCNelson, 2018-05-17:
Sounds about right, I'm guessing your not using the RT Kernel?

vinifr, 2018-05-17:
No, i am not. But I dont have problems with 200uS interval. The problem is the function omap2_mcspi_transfer_one is not called by kernel. I know because I inserted a GPIO toggle inside this function. It is the third signal in images above.

closed 2018-05-30 by vinifr