beagleboard/linux #175: Enabling pins/ports (uart) on Beaglebone Black A5C with Ubuntu 18.04

BBB-noob, 2018-07-11:
Hi!
After many attempts with the various answers already given, I have to resign myself and ask the community for help.

My project uses Ubuntu 18.04 with the kernel 4.14.44-ti-r51 on a Beaglebone Black A5C. Ultimately I am working on ROS but my first goal is to make the pins work properly first, especially by enabling the uart ports for example. I tried the config-pin method, tuning the boot/uEnv.txt file to enable them when booting, loading the beaglebone-universal-io from https://github.com/cdsteinkuehler/beaglebone-universal-io... It seems that in all cases, the pinmux file of the pin I'm trying to change is not found (and I tried several) and that I should verify your device tree file.

I had the same issue on Ubuntu 16.04 and got rid of these issues by switching to Debian 9.3. But now I really need to use Ubuntu (ROS-related) so I would like to have the same possibilities in terms of pins-controlling as with Debian 9.3.

I tried many updates on the file /boot/uEnv.txt but every time, the same error occurs, when I try to set the ports to uart, it give me that:
ubuntu@arm:~$ config-pin P9.11 uart
P9_11 pinmux file not found!
Please verify your device tree file

Right now, /bootuEnv.txt looks like:
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.14.44-ti-r51
#uuid=
#dtb=

##BeagleBone Black/Green dtb's for v4.1.x (BeagleBone White just works..)

##BeagleBone Black: HDMI (Audio/Video) disabled:
#dtb=am335x-boneblack-emmc-overlay.dtb

##BeagleBone Black: eMMC disabled:
#dtb=am335x-boneblack-hdmi-overlay.dtb

##BeagleBone Black: HDMI Audio/eMMC disabled:
#dtb=am335x-boneblack-nhdmi-overlay.dtb

##BeagleBone Black: HDMI (Audio/Video)/eMMC disabled:
dtb=am335x-boneblack-overlay.dtb

##BeagleBone Black: wl1835
#dtb=am335x-boneblack-wl1835mod.dtb

##BeagleBone Green: eMMC disabled
#dtb=am335x-bonegreen-overlay.dtb

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
uboot_overlay_addr0=/lib/firmware/BB-BELA-B2.dtbo
uboot_overlay_addr4=/lib/firmware/BB-UART1-00A0.dtbo
uboot_overlay_addr1=/lib/firmware/BB-UART2-00A0.dtbo
uboot_overlay_addr2=/lib/firmware/BB-UART4-00A0.dtbo
uboot_overlay_addr3=/lib/firmware/BB-UART5-00A0.dtbo
#uboot_overlay_addr0=/lib/firmware/<file0>.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
disable_uboot_overlay_emmc=1
disable_uboot_overlay_video=1
disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
[... 398 lines cut ...]
[    1.518504] pinctrl-single 44e10800.pinmux: pin PIN48 already requested by 481aa000.serial; cannot claim for 0-0070
[    1.529101] pinctrl-single 44e10800.pinmux: pin-48 (0-0070) status -22
[    1.535706] pinctrl-single 44e10800.pinmux: could not request pin 48 (PIN48) from group nxp_hdmi_bonelt_pins  on device pinctrl-single
[    1.555108] pinctrl-single 44e10800.pinmux: pin PIN96 already requested by 48022000.serial; cannot claim for 4802a000.i2c
[    1.566185] pinctrl-single 44e10800.pinmux: pin-96 (4802a000.i2c) status -22
[    1.573298] pinctrl-single 44e10800.pinmux: could not request pin 96 (PIN96) from group pinmux_i2c1_pins  on device pinctrl-single
dmesg | grep gpio-of-helper
[    1.168742] gpio-of-helper ocp:cape-universal: ready
END

ubuntu@arm:~$ ll /dev/ttyS*
crw--w---- 1 root tty     4, 64 Jul 11 17:38 /dev/ttyS0
crw-rw---- 1 root dialout 4, 65 Jul 11 17:38 /dev/ttyS1
crw-rw---- 1 root dialout 4, 66 Jul 11 17:38 /dev/ttyS2
crw-rw---- 1 root dialout 4, 67 Jul 11 17:38 /dev/ttyS3
crw-rw---- 1 root dialout 4, 68 Jul 11 17:38 /dev/ttyS4
crw-rw---- 1 root dialout 4, 69 Jul 11 17:38 /dev/ttyS5

ubuntu@arm:~$ cat /proc/cmdline 
console=ttyO0,115200n8 bone_capemgr.enable_partno=BB-UART1,BB-UART2,BB-UART4.BB-UART5 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet

Lastly, I tried to display $ cat /sys/devices/platform/bone_capemgr/slots
But the slot file (or whatever this is) does not exist, although I could display something before if I recall correctly.

Thanks a lot in advance for any help you could give me!

RobertCNelson, 2018-07-11:
@BBB-noob 

```bash
kernel:[4.14.44-ti-r51]
device-tree-override:[dtb=am335x-boneblack-overlay.dtb]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_addr0=/lib/firmware/BB-BELA-B2.dtbo]
uboot_overlay_options:[uboot_overlay_addr4=/lib/firmware/BB-UART1-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr1=/lib/firmware/BB-UART2-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr2=/lib/firmware/BB-UART4-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr3=/lib/firmware/BB-UART5-00A0.dtbo]
```

In v4.14.x specifying ^ those disables config-pin..

Regards,

RobertCNelson, 2018-07-11:
@BBB-noob ps i've been posting ROS Melodic Morenia weekly images for awhile now here:

https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#ROS_.28ROS_Melodic_Morenia.29_-_Weekly

Everything is really setup for ROS..

Regards,

BBB-noob, 2018-07-12:
Thanks a lot for the answer. Indeed, the already-setup image works perfectly on BBB. I tried it before but now I realised my sd card was dying and could not be read properly. But it worked with a new one.

I was wondering if there was a similar image with Ubuntu 16.04 and ROS Kinetic?