beagleboard/linux #176: Implementing 1 Wire Overlay in BBB - (ds18b20 sensor)

sherCon221b, 2018-07-24:
I'm trying to use ds18b20 in BBB. I used  uboot_overlay_addr0=/lib/firmware/BB-W1-P9.12-00A0.dtbo

with the following software details in my BBB:
***********************************
Static hostname: beaglebone
         Icon name: computer
        Machine ID: e21f1c43d4398ceebe8e92a85b266813
           Boot ID: 4f4a254c9d7c42d1a22c57ca0fd8b44d
  Operating System: Debian GNU/Linux 9 (stretch)
            Kernel: Linux 4.14.49-ti-r54
      Architecture: arm
********************************************************************************

I can detect the device but with no values for a few seconds and then it disappears. 
Below is the output response:
*********************************************************************************
debian@beaglebone:~$ cat /sys/bus/w1/devices/28-000004ef20ed/w1_slave
00 00 00 00 00 00 00 00 00 : crc=00 NO
00 00 00 00 00 00 00 00 00 t=0
debian@beaglebone:~$ cat /sys/bus/w1/devices/28-000004ef20ed/w1_slave
00 00 00 00 00 00 00 00 00 : crc=00 NO
00 00 00 00 00 00 00 00 00 t=0
debian@beaglebone:~$ cat /sys/bus/w1/devices/28-000004ef20ed/w1_slave
00 00 00 00 00 00 00 00 00 : crc=00 NO
00 00 00 00 00 00 00 00 00 t=0
debian@beaglebone:~$ cat /sys/bus/w1/devices/28-000004ef20ed/w1_slave
cat: /sys/bus/w1/devices/28-000004ef20ed/w1_slave: No such file or directory
*********************************************************************************

Is this cause by the software or hardware? 
I hope you could help me with this issue.
THANKS in advance!

cjritola, 2018-10-27:
Bump. I'm in the same boat, except I'm not even lucky enough to have any device show up in the devices/ directory despite doing what sherCon221b did.

c-embed, 2019-02-26:
Hello, I am facing the same issue on a PocketBeagle.
Kernel: 4.14.71-ti-r80
Debian 9.5

I have modified BB-W1-P9.12-00A0.dts, recompiled, and can see the device.
It then fails CRC, and after a while disappears.

Was there ever a resolution to this? Am I right to understand that kernel version 4.18.0 doesn't have this problem?

cjritola, 2019-02-26:
c-embed:

I think you configured things correctly and there's a limitation in the w1-gpio driver. Adding extra pullup resistors helped a little however it was far from perfect. When the Beaglebone's temperature dropped below about 40 degrees, onewire reception became 1:20 poor. When the Beaglebone warmed up it improved to 1:2. The sensor would show up and disappear, sometimes with the incorrect unique ID. 

The readings would often be garbage, usually failing CRC though some bad readings actually got through the CRC check!

I got similar problems using a raspberry pi (similar processor), with a completely different cabling scheme, different wiring style for the resistors, using external pullup instead of internal. Same behavior.

What might be happening is the beaglebone's clock is being affected by the cold, coupled with inflexibilities in the w1-gpio code, causing timing problems and failed parsing.

The GPIO route ended up being _too_ cheap. The solution was to buy a Maxim DS9490R USB-to-1wire dongle. It ended up being a drop-in replacement for w1-gpio - file trees were the same. Signaling has been flawless since. It is pricey for what it is but frankly the hours wasted were worth more.

c-embed, 2019-02-27:
cjritola:

Thank you for your response. Unfortunately, for space and cost constraints, the USB to 1-wire converter will not be an option. I plan on trying this with the updated kernel, to see if this bypasses the issue.

c-embed, 2019-03-01:
In case anyone else has faced this issue:

I updated the kernel to 4.19, and made the above modifications to the .dts file and uEnv.txt. The DS18B20 sensor is shown under /sys/bus/w1/devices, and I am able to read the temperature setting from it. It passes CRC, as well.

I followed the reference (in this thread) to RobertCNelson and dlech's commits, to infer that this would be a possible fix. 

I consider this issue resolved.

closed 2020-03-02 by sherCon221b