beagleboard/linux #177: iio not loaded automatically on Blue

jadonk, 2018-08-09:
```sh
debian@beaglebone:/var/lib/cloud9$ rc_test_adc
ERROR in rc_adc_init, failed to open iio adc interface
: No such file or directory
Perhaps kernel or device tree is too old
ERROR: failed to run rc_init_adc()
```

version.sh output:
```
git:/opt/scripts/:[4a61888e8e27158bcbb0ca70531038d225a8d7af]
eeprom:[A335BNLTBLA21708EL000277]
model:[TI_AM335x_BeagleBone_Blue]
dogtag:[BeagleBoard.org Debian Image 2018-08-05]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.03-00002-gac9cce7c6a]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2017.07-rc1-00002-g4d87f7]:[location: dd MBR]
kernel:[4.14.61-ti-r68]
nodejs:[v6.14.3]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20180803.0-0rcnee0~stretch+20180804]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.0-git20180726.0-0rcnee0~stretch+20180727]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.043124] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
```

jadonk, 2018-08-09:
Verified enabling tscadc in the device tree fixes the issue.

RobertCNelson, 2018-11-19:
@jadonk u-boot is too old on eMMC, adc was switched to an overlay. (at the time librobotcontrol mmap'ed the adc).. That's all now been fixed, yet u-boot needs to load things today..

Regards,