beagleboard/linux #180: error: implicit declaration of function ‘kfree’

ledav-net, 2018-08-27:
Hi,

I've the following issue while trying to build the 4.14.65-ti-r72 release:

15:23:07 In file included from ./include/linux/clocksource.h:19:0,
15:23:07 from ./arch/arm/include/asm/arch_timer.h:7,
15:23:07 from arch/arm/vdso/vgettimeofday.c:21:
15:23:07 ./include/linux/of.h: In function ‘__of_changeset_add_update_property_stringv’:
15:23:07 ./include/linux/of.h:1455:2: error: implicit declaration of function ‘kfree’ [-Werror=implicit-function-declaration]
15:23:07 kfree(str);
15:23:07 ^~~~~
15:23:07 cc1: some warnings being treated as errors
13:23:07 make[3]: *** [scripts/Makefile.build:329: arch/arm/vdso/vgettimeofday.o] Error 1
15:23:07 make[2]: *** [Makefile:1035: arch/arm/vdso] Error 2
15:23:07 make[2]: *** Waiting for unfinished jobs....
15:23:07 CC      init/do_mounts_initrd.o
15:23:07 CC      init/initramfs.o
15:23:08 CC      arch/arm/kernel/process.o
15:23:08 CC      arch/arm/kernel/ptrace.o
15:23:08 CC      init/calibrate.o
15:23:09 CC      init/init_task.o
15:23:09 CC      init/version.o
15:23:09 In file included from ./include/linux/clocksource.h:19:0,
15:23:09 from ./include/linux/clockchips.h:14,
15:23:09 from ./include/linux/tick.h:8,
15:23:09 from arch/arm/kernel/process.c:28:
15:23:09 ./include/linux/of.h: In function ‘__of_changeset_add_update_property_stringv’:
15:23:09 ./include/linux/of.h:1455:2: error: implicit declaration of function ‘kfree’; did you mean ‘kvfree’? [-Werror=implicit-function-declaration]
15:23:09 kfree(str);
15:23:09 ^~~~~
15:23:09 kvfree
15:23:09 cc1: some warnings being treated as errors
15:23:09 make[3]: *** [scripts/Makefile.build:328: arch/arm/kernel/process.o] Error 1
15:23:09 make[3]: *** Waiting for unfinished jobs....
15:23:09 AR      init/mounts.o
15:23:09 AR      init/built-in.o
15:23:09 make[2]: *** [Makefile:1035: arch/arm/kernel] Error 2

I fixed it by adding an "#include <linux/slab.h>" on top of include/linux/of.h.

Regards,
David.

mvduin, 2018-09-24:
Same problem in 4.14.71-bone16

RobertCNelson, 2018-09-24:
Thanks @ledav-net and @mvduin another bug i completely lost track off..

@mvduin it'll hit the bone repo shortly, just tagging the ti branch and pushing that one out first..

Regards,

closed 2018-09-24 by RobertCNelson

scharri, 2019-05-22:
I've tried to build "linux 4.9.147-ti-r120" and got the same error as @ledav-net .
Maybe we should patch this issue and apply it to the mainline kernels?