beagleboard/linux #189: no /dev/fb0 in 4.19 branch

lvd2, 2018-12-19:
I'm using devuan ascii distribution image and compile this 4.19 branch using .config taken from distribution kernel (4.9.130). Board is beaglebone black.

lvd@devuan ~ $ ls /dev/fb*
ls: cannot access '/dev/fb*': No such file or directory

The resulting dmesg and config.gz are in attaches

[config.gz](https://github.com/beagleboard/linux/files/2695423/config.gz)
[dmesg.gz](https://github.com/beagleboard/linux/files/2695424/dmesg.gz)

antailbaxt3r, 2019-01-02:
Could you try using the command "ls dev/fb" instead of "ls /dev/fb" and tell us the output then please?

lvd2, 2019-01-03:
I've tried actually connecting HDMI monitor to the board. The /dev/fb0 appeared with the login prompt on the screen.

closed 2019-01-03 by lvd2