beagleboard/linux #191: Why can only "nxp,tda998x"  be used ?

Pillar1989, 2019-01-31:
https://github.com/beagleboard/linux/blob/83521bea4e860954320fc0ee491c7e2084f22aec/drivers/gpu/drm/tilcdc/tilcdc_external.c#L237-L240

A lot of people want to use SeeedStudio's high cost performance HDMI cape 
 https://www.seeedstudio.com/SeeedStudio-BeagleBone-Green-HDMI-Cape-p-2570.html  on the latest kernel, but because of this line of code, is not available

pdp7, 2020-06-09:
@Pillar1989 are you using a different HDMI chip than tda998x?

@RobertCElliott @jadonk is tda998x the only HDMI driver that is supported by the

Pillar1989, 2020-06-11:
@pdp7  The Linux kernel supports many LCD to HDMI chips. Obviously, it doesn't make sense to write code like this.

pdp7, 2020-06-11:
@Pillar1989 I discussed this with @RobertCNelson and he pointed out there is the Arrow BBB Industrial that uses adv7511 HDMI chip:
https://github.com/beagleboard/BeagleBoard-DeviceTrees/blob/4a9c0a652f58090491319d27dac4bf76da7d6086/src/arm/am335x-abbbi.dts#L91

The "adv7511" is a proper "bridge" driver: https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/gpu/drm/bridge
4:30

While the "tda998x_drv.c" is a special... https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/gpu/drm/i2c
4:31
> The tda998x should be converted to a bridge driver, but no one's cared enough to change that.  It's used on a lot of different arm boards.

Looking at the Seeed cape, I see it is using IT66121 HDMI Framer:
https://wiki.seeedstudio.com/BeagleBone_Green_HDMI_Cape/

There was a patch to add it66121 driver to mainline posted by Phong LE in March 2020 that might be useful:
https://lwn.net/Articles/814501/

In summary,  tilcdc_get_external_components() in mainline does appear to be correct in that it is handling the the fact that tda998x is old and not covered by a real bridge driver.

Pillar1989, 2020-06-15:
Yes, I see. Thank you

closed 2020-06-15 by Pillar1989