beagleboard/linux #192: Kernel config build files mismatch between releases

AnAccountForReportingBugs, 2019-03-07:
When I updated from kernel 4.4.113-ti-r149 to 4.4.155-ti-r151 my BBB running an IOT image didn't boot properly.  I could not get HDMI output to access the emergency console.  Instead, after replacing my F2FS SD card with a new one from a nearby store running the latest official IOT image, I was able to boot from the SD card and access it over USB.  When looking at the logs, the problem was traced to the SD card partition in the fstab without the `nofail` option and the fact that in the r151 kernel `# CONFIG_F2FS_FS is not set` and thus the mount phase failed, triggering emergency mode.

My recommended solution is to have some sort of unified baseline kernel config for most features to prevent such surprises in the future across updates / upgrades.

I've included a the output of `diff -y --suppress-common-lines /boot/config*` and version.sh for your reference.

Thank you for your attention in this manner.

[149-151.diff.txt](https://github.com/beagleboard/linux/files/2943509/149-151.diff.txt)

git:/opt/scripts/:[247665d553b6b28fdd47c38e80e910a9208772be]
eeprom:[A335BNLT000C4414BBBK0752]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2017-03-19]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2017.03-00002-gd12b1519b4]:[location: dd MBR]
kernel:[4.4.113-ti-r149]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190227.0-0rcnee0~jessie+20190227]
WARNING:pkg:[bb-wl18xx-firmware]:[NOT_INSTALLED]
pkg:[kmod]:[18-3rcnee1~jessie+20171005]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
WARNING:pkg:[firmware-ti-connectivity]:[NOT_INSTALLED]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal netdev i2c bluetooth admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 root=UUID=9d270fd5-959b-4d3e-976d-648e14049a7a ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet cape_universal=enable fsck.mode=force]
dmesg | grep pinctrl-single
[    3.192226] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
[   41.450534] pinctrl-single 44e10800.pinmux: pin 44e109ac.0 already requested by 48038000.mcasp; cannot claim for ocp:P9_25_pinmux
[   41.462405] pinctrl-single 44e10800.pinmux: pin-107 (ocp:P9_25_pinmux) status -22
[   41.469996] pinctrl-single 44e10800.pinmux: could not request pin 107 (44e109ac.0) from group pinmux_P9_25_default_pin  on device pinctrl-single
[   41.544849] pinctrl-single 44e10800.pinmux: pin 44e109ac.0 already requested by 48038000.mcasp; cannot claim for ocp:P9_25_pinmux
[   41.556686] pinctrl-single 44e10800.pinmux: pin-107 (ocp:P9_25_pinmux) status -22
[   41.564269] pinctrl-single 44e10800.pinmux: could not request pin 107 (44e109ac.0) from group pinmux_P9_25_default_pin  on device pinctrl-single
[   41.684756] pinctrl-single 44e10800.pinmux: pin 44e1099c.0 already requested by 48038000.mcasp; cannot claim for ocp:P9_28_pinmux
[   41.696611] pinctrl-single 44e10800.pinmux: pin-103 (ocp:P9_28_pinmux) status -22
[   41.704189] pinctrl-single 44e10800.pinmux: could not request pin 103 (44e1099c.0) from group pinmux_P9_28_default_pin  on device pinctrl-single
[   41.789647] pinctrl-single 44e10800.pinmux: pin 44e1099c.0 already requested by 48038000.mcasp; cannot claim for ocp:P9_28_pinmux
[   41.801508] pinctrl-single 44e10800.pinmux: pin-103 (ocp:P9_28_pinmux) status -22
[   41.809095] pinctrl-single 44e10800.pinmux: could not request pin 103 (44e1099c.0) from group pinmux_P9_28_default_pin  on device pinctrl-single
[   41.899526] pinctrl-single 44e10800.pinmux: pin 44e10994.0 already requested by 48038000.mcasp; cannot claim for ocp:P9_29_pinmux
[   41.911459] pinctrl-single 44e10800.pinmux: pin-101 (ocp:P9_29_pinmux) status -22
[   41.919020] pinctrl-single 44e10800.pinmux: could not request pin 101 (44e10994.0) from group pinmux_P9_29_default_pin  on device pinctrl-single
[   42.021643] pinctrl-single 44e10800.pinmux: pin 44e10994.0 already requested by 48038000.mcasp; cannot claim for ocp:P9_29_pinmux
[   42.033502] pinctrl-single 44e10800.pinmux: pin-101 (ocp:P9_29_pinmux) status -22
[   42.041092] pinctrl-single 44e10800.pinmux: could not request pin 101 (44e10994.0) from group pinmux_P9_29_default_pin  on device pinctrl-single
[   42.156801] pinctrl-single 44e10800.pinmux: pin 44e10990.0 already requested by 48038000.mcasp; cannot claim for ocp:P9_31_pinmux
[   42.168618] pinctrl-single 44e10800.pinmux: pin-100 (ocp:P9_31_pinmux) status -22
[   42.176217] pinctrl-single 44e10800.pinmux: could not request pin 100 (44e10990.0) from group pinmux_P9_31_default_pin  on device pinctrl-single
[   42.249684] pinctrl-single 44e10800.pinmux: pin 44e10990.0 already requested by 48038000.mcasp; cannot claim for ocp:P9_31_pinmux
[   42.261538] pinctrl-single 44e10800.pinmux: pin-100 (ocp:P9_31_pinmux) status -22
[   42.269122] pinctrl-single 44e10800.pinmux: could not request pin 100 (44e10990.0) from group pinmux_P9_31_default_pin  on device pinctrl-single
dmesg | grep gpio-of-helper
[    3.193055] gpio-of-helper ocp:cape-universal: Allocated GPIO id=0
[    3.193213] gpio-of-helper ocp:cape-universal: Allocated GPIO id=1
[    3.193224] gpio-of-helper ocp:cape-universal: ready
END

RobertCNelson, 2019-03-08:
Fixed this last night:

https://github.com/RobertCNelson/ti-linux-kernel-dev/commit/78dc8dbc2610a7aa65c856e9812f80a6b16e070f#diff-ba79210cd3316674807a5388d265a3f2

I'll re-enable that shortly in the v4.9.x/v4.14.x/v4.19.x branches..

Regards,

AnAccountForReportingBugs, 2019-03-08:
I appreciate the swift attention to my issue.  I've not tried upgrading kernels yet, but that looks like it should work (at least the part I can understand).  I will leave this open until I actually try the kernel, but feel free to close it early if you don't hear from me in a time you consider reasonable.

I can't help but think that there needs to be some sort of baseline kernel configuration for the BBB in terms of enabled (either built in or as a module) file systems and hardware.  Maybe there is, I don't know about it, and F2FS use is now rarer than I thought.  One of these days, maybe I'll muster up the bravery to compile my own kernel on the BBBs.  Regardless, I again thank you for your attention in this matter.

AnAccountForReportingBugs, 2019-03-22:
I've updated to the new kernel and the F2FS driver is working, so my instant problem is fixed.  Another change of note is that now my kernel ring buffer is full of stack traces from `bone-pinmux-helper` and `pinctrl-single` in addition to the error messages in my original post.  Since, I'm pretty sure you are already aware of them, I will, therefore, not consider them to be an issue.

closed 2019-03-22 by AnAccountForReportingBugs