beagleboard/linux #194: no gfs2 in 4.14 branch?

moosbaue, 2019-03-16:


RobertCNelson, 2019-03-18:
@moosbaue, sorry about that, it got removed in a fs cleanup, i'll add CFS2 as a hard default, should be pushed out to all kernels by mid week. (v4.4.x+ -> 5.1.0-rc1)

```code
CONFIG_GFS2_FS=m
CONFIG_GFS2_FS_LOCKING_DLM=y
```

Regards,

moosbaue, 2019-03-18:
> Reply-to: beagleboard/linux <reply+00057d36f286aca5df3cce547e532b156a609fc738ab111b92cf0000000118a6
> b45892a169ce19249d05@reply.github.com>

Thank you Nelson

Saves me from (cross)building kernels myself
I appreciate this very much

kind regards
Josef

tity":{"external_key":"github/beagleboard/linux","title":"beagleboard/linux","subtitle":"GitHub
repository","main_image_url":"https://github.githubassets.com/images/email/message_cards/header.png
","avatar_image_url":"https://github.githubassets.com/images/email/message_cards/avatar.png","actio
n":{"name":"Open in
GitHub","url":"https://github.com/beagleboard/linux"}},"updates":{"snippets":[{"icon":"PERSON","mes
sage":"@RobertCNelson in #194: @moosbaue, sorry about that, it got removed in a fs cleanup, i'll
add CFS2 as a hard default, should be pushed out to all kernels by mid week. (v4.4.x+ -\u003e
5.1.0-
rc1)\r\n\r\n```code\r\nCONFIG_GFS2_FS=m\r\nCONFIG_GFS2_FS_LOCKING_DLM=y\r\n```\r\n\r\nRegards,\r\n"
}],"action":{"name":"View
Issue","url":"https://github.com/beagleboard/linux/issues/194#issuecomment-473737289"}}}
[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/beagleboard/linux/issues/194#issuecomment-473737289",
"url": "https://github.com/beagleboard/linux/issues/194#issuecomment-473737289",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

kind regards / mit freundlichen Grüßen

Moosbauer Josef

<josef@moosbauer.net>

GPG Fingerprint 6748 9413 0EF1 5B23 D707  DBFA 570B 00E3 4689 8CC9

"Of course I'm crazy, but that doesn't mean I'm wrong.I'm mad but not ill"

closed 2019-03-18 by moosbaue