beagleboard/linux #195: 3.8.13- version bone73 does not support Kingstom eMMC

flightxgf, 2019-03-19:
Hello!
The eMMC used by beaglebone black here is Kingstom eMMC, and the kernel version is 3.8.13-bone73. 
The kernel starts up, and devices without the Kingstom eMMC, such as /dev/mmcblk1, are enabled,There are no mmcblk1 devices in the directory /sys/block/.

The latest image I try to use https://beagleboard.org/latest-images, found that can support Kingstom eMMC,

My question now is what to do in the 3.8.13-bone73 kernel version to support Kingstom eMMC device support, you can see the device name such as /dev/mmcblk1 and /sys/block/mmcblk1

Thank you very much for your reply. Looking forward to your reply.

Best regards,

RobertCNelson, 2019-03-19:
@flightxgf 

Congrats on still using an ancient kernel and not having your system hacked! This issue was first fixed with:

https://github.com/RobertCNelson/bb-kernel/commit/bd5a1f38c99bbe712d21c1e145b0c660dca9dee7

on: Jun 15, 2016

and released as 3.8.13-bone80, thus, to modify your old un-supported, insecure, old kernel follow this Procedure

https://gist.github.com/RobertCNelson/39faf80ddc9fcefae74dce2c6ca2eb45#file-procedure

Regards,

flightxgf, 2019-03-20:
Thank you so much for @robertcnelson's reply

closed 2019-03-20 by flightxgf