beagleboard/linux #196: 插入SD卡打印错误信息

flightxgf, 2019-03-20:
The hardware is BeagleBone Black，System image is https://beagleboard.org/latest-images, (the use of bone-debian-9.5-iot-armhf-2018-10-07-4gb).

After the images is burned, start the system and enter the terminal. When I insert the usb flash drive(Kinston 32G), The terminal prints the following information. It prints about once a second.

```
debian@beaglebone:~$ [  142.971926] print_req_error: I/O error, dev sda, sector 60631040
[  144.740522] print_req_error: I/O error, dev sda, sector 60631160
[  150.032573] print_req_error: I/O error, dev sda, sector 60631480
[  152.384743] print_req_error: I/O error, dev sda, sector 60631552
[  154.144554] print_req_error: I/O error, dev sda, sector 60631672

debian@beaglebone:~$ [  191.736260] sd 0:0:0:0: [sda] No Caching mode page found
[  191.745822] sd 0:0:0:0: [sda] Assuming drive cache: write through
[  195.124878] print_req_error: I/O error, dev sda, sector 60631552
[  196.884366] print_req_error: I/O error, dev sda, sector 60631672
[  200.904439] print_req_error: I/O error, dev sda, sector 60630912
[  203.868704] print_req_error: I/O error, dev sda, sector 60631168
```
Do I need to patch it or can it be solved by other ways?

Looking forward to your guidance. Thank you very much for your help
Best regards,

RobertCNelson, 2019-03-20:
@flightxgf , how are you powering the board? the 5Vdc jack or USB.. If it's USB, try the 5Vdc jack, otherwise your usb flash drive looks shot...

flightxgf, 2019-03-21:
> @flightxgf , how are you powering the board? the 5Vdc jack or USB.. If it's USB, try the 5Vdc jack, otherwise your usb flash drive looks shot...

Thank you very much for your reply.

I use the 5Vdc. 

The BeagleBone Black is not connected to anything except a usb flash drive (Kinston 32G) and 5vdc and hdmi.

Today use the latest official mirror https://beagleboard.org/latest-images, (the use of ipads - debian - 9.5 - the iot - armhf - 2018-10-07-4 gb). The test found that the problem is
debian@beaglebone:~$ [  142.971926] print_req_error: I/O error, dev sda, sector 60631040
```
[  144.740522] print_req_error: I/O error, dev sda, sector 60631160
[  150.032573] print_req_error: I/O error, dev sda, sector 60631480
[  152.384743] print_req_error: I/O error, dev sda, sector 60631552
[  154.144554] print_req_error: I/O error, dev sda, sector 60631672

debian@beaglebone:~$ [  191.736260] sd 0:0:0:0: [sda] No Caching mode page found
[  191.745822] sd 0:0:0:0: [sda] Assuming drive cache: write through
[  195.124878] print_req_error: I/O error, dev sda, sector 60631552
[  196.884366] print_req_error: I/O error, dev sda, sector 60631672
[  200.904439] print_req_error: I/O error, dev sda, sector 60630912
```
Looking forward to your guidance. Thank you very much for your help
Best regards,

closed 2019-03-22 by flightxgf