beagleboard/linux #197: The device on which the BOOT button was not found

flightxgf, 2019-03-20:
Hello,The Linux version I use is 3.8.13-bone73.The hardware is beaglebone black.
When I power on and press the BOOT BUTTON BUTTON, I can select SD and eMMC to start the system, but when I enter the system, the name of the BOOT BUTTON device cannot be seen under /dev/ and /sys /devices.

I want to know whether the system works when I press and lift the BOOT BUTTON after the system is up. How can I find the information like /dev/s2boot and /ses/devices? 

Looking forward to your guidance. Thank you very much for your help
Best regards,

RobertCNelson, 2019-03-20:
@flightxgf , the Boot Button is tied to a "P8.43" on the header, just export that pin and check the value your self...

Regards,

flightxgf, 2019-03-21:
> @flightxgf , the Boot Button is tied to a "P8.43" on the header, just export that pin and check the value your self...
> Regards,

Thanks.
How do I export it?

Looking forward to your guidance. Thank you very much for your help
Regards,

flightxgf, 2019-03-22:
> @flightxgf , the Boot Button is tied to a "P8.43" on the header, just export that pin and check the value your self...
> Regards,

flightxgf, 2019-03-22:
> @flightxgf , the Boot Button is tied to a "P8.43" on the header, just export that pin and check the value your self...
> Regards,
@RobertCNelson 
Hello, Robert, when the system is up, I tried all kinds of methods but could not get the value of the p8.43(SYS_BOOT) button, may I ask what I should do? Thanks。

closed 2019-03-22 by flightxgf