beagleboard/linux #201: PocketBeagle floating inputs should have RX disabled

jadonk, 2019-04-06:
The pinmux for P1.2 and P2.35 should be set to RX disabled for the case where an analog input is provided on those pins floating between 0V and 3.3V. Otherwise, there is extra I/O cell leakage.

http://processors.wiki.ti.com/index.php/AM335x_Schematic_Checklist

pdp7, 2020-06-09:
@jadonk do you mean disabled by kernel driver?

pdp7, 2020-07-08:
@jadonk I think you mean setting the mux mode to have RX disabled in the device tree
Per [the cape header pins sheet](https://docs.google.com/spreadsheets/d/1fE-AsDZvJ-bBwzNBj1_sPDrutvEvsmARqFwvbw_HkrE/edit#gid=1518010293), I believe `P1.02 / AIN6 / GPIO87` is:
- mode0: lcd hsync
- mode7: gpio 2.23

and `P2.35 / AIN5 / GPIO86` is:
- mode0: lcd vsync
- mode7: gpio 2.22

Does that look right?