beagleboard/linux #202: 1-wire overlay in BBG, Debian, Kernel 4.14

JeanAnouilh, 2019-04-10:
Hello Everybody,

I want to enable e specific pin to be used as a 1-wire bus, but everything I've tried didn't work. There were neither errors nor was there ever any device in /sys/bus/w1/devices/.

Here are some information about the system I use:

`sudo /opt/scripts/tools/version.sh`
`git:/opt/scripts/:[e9bcff232834702c1c810710706ee815d77b080b]
eeprom:[A335BNLTBBG1BBG116045614]
model:[TI_AM335x_BeagleBone_Green]
dogtag:[BeagleBoard.org Debian Image 2018-10-14]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-gd5b4c4b656]:[location: dd MBR]
kernel:[4.14.71-ti-r80]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_addr1=/lib/firmware/BB-UART2-00A0.dtbo]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[disable_uboot_overlay_wireless=1]
uboot_overlay_options:[disable_uboot_overlay_adc=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
uboot_overlay_options:[uboot_overlay_addr5=/lib/firmware/BB-PPS-P9_12-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr0=/lib/firmware/BB-W1-P8.11-00A0.dtbo]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190404.0-0rcnee0~stretch+20190404]
pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~stretch+20190227]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal netdev gpio pwm eqep admin spi i2c tisdk weston-launch xenomai bluetooth cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.119468] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.131469] gpio-of-helper ocp:cape-universal: ready
END`

`cat /boot/uEnv.txt`
`#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.14.71-ti-r80
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
#uboot_overlay_addr0=/lib/firmware/BB-UART1-00A0.dtbo
uboot_overlay_addr1=/lib/firmware/BB-UART2-00A0.dtbo
#uboot_overlay_addr2=/lib/firmware/BB-UART3-00A0.dtbo
#uboot_overlay_addr3=/lib/firmware/BB-UART4-00A0.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
disable_uboot_overlay_video=1
disable_uboot_overlay_audio=1
disable_uboot_overlay_wireless=1
disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_rproc (4.14.x-ti kernel)
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo
###pru_uio (4.4.x-ti, 4.14.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks... (60000 = 384KB)
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 quiet
nohz=off

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 quiet video=HDMI-A-1:1024x768@60e

#Use an overlayfs on top of a read-only root filesystem:
#cmdline=coherent_pool=1M net.ifnames=0 quiet overlayroot=tmpfs

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh

###Additional custom capes
uboot_overlay_addr5=/lib/firmware/BB-PPS-P9_12-00A0.dtbo
uboot_overlay_addr0=/lib/firmware/BB-W1-P8.11-00A0.dtbo`

And this is the .dts file I've used:
`/*
* Copyright (C) 2012 Texas Instruments Incorporated - http://www.ti.com/
*
* This program is free software; you can redistribute it and/or modify
* it under the terms of the GNU General Public License version 2 as
* published by the Free Software Foundation.
* 
* Modified  by Russell Senior from the weather cape's DTS file.
* Minor formatting by C W Rose.
*/
/dts-v1/;
/plugin/;
 
/ {
    compatible = "ti,beaglebone", "ti,beaglebone-black";
    part-number = "BB-W1-P8.11";
    version = "00A0";
 
    exclusive-use = "P8.11";
 
    fragment@0 {
        target = <&am33xx_pinmux>;
        __overlay__ {
             bb_w1_pins: pinmux_bb_w1_pins {
                 pinctrl-single,pins =  <0x34 0x37 /* gpmc_ad13.gpio1_13, OMAP_PIN_INPUT_PULLUP | OMAP_MUX_MODE7 - w1-gpio */ >;
             };
        };
    };
 
    fragment@1 {
        target = <&ocp>;
        __overlay__ {
            onewire@0 {
                status          = "okay";
                compatible      = "w1-gpio";
                pinctrl-names   = "default";
                pinctrl-0       = <&bb_w1_pins>;
 
                gpios = <&gpio1 13 0>;
            };
        };
    };
};`

I really don't know what I'm doing wrong that it's not working.

Thanks for any help.

RobertCNelson, 2019-04-10:
@JeanAnouilh onewire isn't under ocp:

```
    fragment@1 {
        target = <&ocp>;
        __overlay__ {
            onewire@0 {
```

It's under /

https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/BB-W1-P9.12-00A0.dts#L53

```
	fragment@2 {
		target-path="/";
		__overlay__ {

			onewire {
```

Regards,

JeanAnouilh, 2019-04-11:
@RobertCNelson Thanks a lot, but now I'm getting this error:
`Error: w1.dts:16.1-9 syntax error
FATAL ERROR: Unable to parse input tree`

when I use:
`dtc -O dtb -o w1-00A0.dtbo -b 0 -@ w1.dts`

RobertCNelson, 2019-04-11:
@JeanAnouilh 

> when I use:
> `dtc -O dtb -o w1-00A0.dtbo -b 0 -@ w1.dts`

Sorry, I'm not going to debug that, stick the file under ./src/arm/ and run make..

> When I remove the #include lines and replace some parts with hardcode the compilation works, but then again I don't see a device in /sys/bus/w1/devices/.
> [w1.dts.txt](https://github.com/beagleboard/linux/files/3067590/w1.dts.txt)

github says: "Not Found"

see example:

https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/BB-W1-P9.12-00A0.dts

Regards,

closed 2019-04-11 by RobertCNelson

JeanAnouilh, 2019-04-11:
> Sorry, I'm not going to debug that, stick the file under ./src/arm/ and run make..

You don't have to be sorry, I didn't expect that. But thanks for the hint with make, this perfectly worked. That was all I was missing the entire time.