beagleboard/linux #204: Cannot configure OneWire or UART for on bone-debian-9.5-iot

kwijbrans, 2019-05-05:
I am trying to get my BBB to work with a OneWire interface a P9.12 and UART4 enabled. Added uboot_overlay_addr0=/lib/firmware/BB-UART4-00A0.dtbo
uboot_overlay_addr1=/lib/firmware/BB-W1-P9.12-00A0.dtbo

Somehow, it does not work: I do not see any devices on the OneWire bus even though two were present when I used Debian Jessie, and I do not see the UART4 device /dev/ttyO4.

sudo /opt/scripts/tools/version.sh gives:
klaas@beaglebone:~$ sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[1aa73453b2c980b75e31e83dab7dd8b6696f10c7]
eeprom:[A335BNLT0A5C2513BBBK0336]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-10-07]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-g0b54a51eee]:[location: dd MBR]
kernel:[4.14.71-ti-r80]
nodejs:[v6.14.4]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
uboot_overlay_options:[uboot_overlay_addr0=/lib/firmware/BB-UART4-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr1=/lib/firmware/BB-W1-P9.12-00A0.dtbo]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20180928.0-0rcnee0~stretch+20180928]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.3-git20181005.0-0rcnee0~stretch+20181005]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0]
dmesg | grep pinctrl-single
[    1.039846] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.041080] gpio-of-helper ocp:cape-universal: ready
END

What can be wrong? Output seems very similar to output of the working configuration in issue #142.

RobertCNelson, 2019-05-05:
@kwijbrans , okay you have an issue another user posted a few week back..

So verison.sh is only detecting the version of u-boot on teh microSD drive:

```
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-g0b54a51eee]:[location: dd MBR]
k
```

and from the cmdline, we see a special flag isn't being passed.. 

```
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0]
```

Thus, you have "something" on the eMMC, can you please figure that out and share with me the "flasher" that programmed your eMMC?  Then i can update version.sh to detect this issue and report it for the next user. (Technically you are the "2nd" user, but the "1st" user with this issue couln't remember anything..) 

Regards,

kwijbrans, 2019-05-06:
@RobertCNelson: Unfortunately, I do not know the flasher that was used previously, this is a system that was configured to also use a harddisk and the harddisk crashed. Trying to get it up and running again.

Is the eMMC on /dev/mmcblk1 when booted from the SD? I have dumped the first 4 MB using dd if=/dev/mmcblk1, would this be of use?

kwijbrans, 2019-05-06:
One addition: I zeroed the eMMC with sudo dd if=/dev/zero of=/dev/mmcblk1 bs=1M count=10 and now it works.

RobertCNelson, 2019-05-07:
@kwijbrans yeah, if you can send me the 4MB *.img i'll do some work on debugging what it was. ;)

kwijbrans, 2019-05-11:
@RobertCNelson: attached the dump of the first 4 MB, hope it helps finding the issue.

[mmcblk1.zip](https://github.com/beagleboard/linux/files/3168907/mmcblk1.zip)

WalterCEden, 2019-05-24:
I may be having the same problem so I'm posting several items below.   I'm also a relatively newbie to BBB and trying to get a DS18B20 temperature sensor working with 1-wire. Here is the output from version.sh (this was booted from an SD card).

git:/opt/scripts/:[1aa73453b2c980b75e31e83dab7dd8b6696f10c7]
eeprom:[A335BNLT00C04417BBBK1847]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-10-07]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-g0b54a51eee]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot2018.03-00002-gac9cce7c6a]:[location: dd MBR]
kernel:[4.14.71-ti-r80]
nodejs:[v6.14.4]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_addr0=/lib/firmware/BB-W1-P9.12-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade ]
pkg:[bb-cape-overlays]:[4.4.20180928.0-0rcnee0stretch+20180928]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0stretch+20180517]
pkg:[kmod]:[23-2rcnee1stretch+20171005]
pkg:[librobotcontrol]:[1.0.3-git20181005.0-0rcnee0stretch+20181005]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[ 1.107595] pinctrl-single 44e10800.pinmux: 142pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[ 1.119429] gpio-of-helper ocp:cape-universal: ready
END

Here is my uEnv.txt file.

#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.14.71-ti-r80
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1

###Overide capes with eeprom
uboot_overlay_addr0=/lib/firmware/BB-W1-P9.12-00A0.dtbo
#uboot_overlay_addr1=/lib/firmware/.dtbo
#uboot_overlay_addr2=/lib/firmware/.dtbo
#uboot_overlay_addr3=/lib/firmware/.dtbo

###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/.dtbo
#uboot_overlay_addr5=/lib/firmware/.dtbo
#uboot_overlay_addr6=/lib/firmware/.dtbo
#uboot_overlay_addr7=/lib/firmware/.dtbo

###Custom Cape
#dtb_overlay=/lib/firmware/w1-00A0.dtbo

###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
#disable_uboot_overlay_video=1
#disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1

###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_rproc (4.14.x-ti kernel)
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo
###pru_uio (4.4.x-ti, 4.14.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo

###Cape Universal Enable
enable_uboot_cape_universal=1

###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1

###U-Boot fdt tweaks... (60000 = 384KB)
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 quiet

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 quiet video=HDMI-A-1:1024x768@60e

#Use an overlayfs on top of a read-only root filesystem:
#cmdline=coherent_pool=1M net.ifnames=0 quiet overlayroot=tmpfs

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh

So I booted by holding down the user button while pressing power on.

I executed the two sudo modprobe commands with no errors report.

All I see in /sys/bus/w1/devices is w1_bus_master1

Suggestions?