beagleboard/linux #208: Adding support for beaglebone compatible platform

tehkillerbee, 2019-07-08:
Hello,

I have been using a platform that is based around the beaglebone platform architecture.
The "core module" is designed by GoEmbed and uses the AM3354 and TI TPS65910A3 PMU. It supports a number of carrier that uses the standard beaglebone cape pinout.
http://www.goembed.com/index.php/Products/detail/tpid/22 
http://www.goembed.com/index.php/Products/detail/tpid/49

The platform is not directly compatible with a standard beaglebone black image due to a sligthly different device tree configuration and due to a slightly different u-boot/MLO. For this reason, it only runs with an outdated debian 8.7. 

Therefore, I would like to add support for it. The original device tree created by GoEmbed is based on a beaglebone black / am335x-evm and sources are provided by GoEmbed. I have already made the necessary changes to the latest am335x-boneblack.dtb and am335x-boneblack-uboot-univ.dtb to get it to boot correctly. But I would like to have a separate dtb since this is not a boneblack :)

1. What is the correct procedure for adding a custom dts? 
2. What is the correct procedure for selecting between a standard beaglebone dts and a custom one in uEnv?

Are there any interest in adding support for this platform?

RobertCNelson, 2019-07-08:
@tehkillerbee , do you know what eeprom value they stuck on these boards?

Please run:
```
sudo /opt/scripts/tools/version.sh
```
For the dtb files, you can submit merge requests to this repo:

https://github.com/beagleboard/BeagleBoard-DeviceTrees

(v4.14.x-ti and v4.19.x-ti branches are most utilzed today)

and they'll get pulled into the current builds..

Regards,

tehkillerbee, 2019-07-09:
@RobertCNelson Thank you. It looks like there is a device at i2c0, address 0x12 (in addition to the TPS at 0x2d). 0x12 appears to be completely empty when I run i2cdump. I am not 100% convinced this is indeed an eeprom. So without an eeprom, it will be impossible to identify the board type at boot, correct? 

When running version.sh, I get the following output:

> debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh
> [sudo] password for debian:
> git:/opt/scripts/:[46d33928c4a396ef472829bcf22761546e550a6f]
> model:[TI_AM335x_BeagleBone_Black]
> dogtag:[BeagleBoard.org Debian Image 2018-10-07]
> bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2015.07-gefd62d2-dirty]:[location: dd MBR]
> kernel:[4.14.108-ti-r108]
> nodejs:[v6.17.0]
> /uEnv.txt exists, uboot overlays is DISABLED, remove /uEnv.txt
> uboot_overlay_options:[enable_uboot_overlays=1]
> uboot_overlay_options:[uboot_overlay_addr4=BB-I2C1-00A0.dtbo]
> uboot_overlay_options:[disable_uboot_overlay_video=1]
> uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
> uboot_overlay_options:[enable_uboot_cape_universal=1]
> pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
> pkg:[bb-cape-overlays]:[4.4.20190610.0-0rcnee0~stretch+20190610]
> pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~stretch+20190227]
> pkg:[kmod]:[23-2rcnee1~stretch+20171005]
> pkg:[librobotcontrol]:[1.0.4-git20190227.1-0rcnee0~stretch+20190327]
> pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
> groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
> cmdline:[console=tty0 console=ttyO0,115200n8 root=/dev/mmcblk0p1 rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet nohz=off]
> dmesg | grep remote
> [    2.438287] remoteproc remoteproc0: wkup_m3 is available
> [    2.519928] remoteproc remoteproc0: powering up wkup_m3
> [    2.519949] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
> [    2.520211] remoteproc remoteproc0: remote processor wkup_m3 is now up
> dmesg | grep pru
> dmesg | grep pinctrl-single
> [    0.959846] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
> [    1.109258] pinctrl-single 44e10800.pinmux: pin PIN56 already requested by panel; cannot claim for ocp:P8_27_pinmux
> [    1.120061] pinctrl-single 44e10800.pinmux: pin-56 (ocp:P8_27_pinmux) status -22
> [    1.127637] pinctrl-single 44e10800.pinmux: could not request pin 56 (PIN56) from group pinmux_P8_27_default_pin  on device pinctrl-single
> [    1.149812] pinctrl-single 44e10800.pinmux: pin PIN58 already requested by panel; cannot claim for ocp:P8_28_pinmux
> [    1.160508] pinctrl-single 44e10800.pinmux: pin-58 (ocp:P8_28_pinmux) status -22
> [    1.168056] pinctrl-single 44e10800.pinmux: could not request pin 58 (PIN58) from group pinmux_P8_28_default_pin  on device pinctrl-single
> [    1.190159] pinctrl-single 44e10800.pinmux: pin PIN57 already requested by panel; cannot claim for ocp:P8_29_pinmux
> [    1.200839] pinctrl-single 44e10800.pinmux: pin-57 (ocp:P8_29_pinmux) status -22
> [    1.208391] pinctrl-single 44e10800.pinmux: could not request pin 57 (PIN57) from group pinmux_P8_29_default_pin  on device pinctrl-single
> [    1.230482] pinctrl-single 44e10800.pinmux: pin PIN59 already requested by panel; cannot claim for ocp:P8_30_pinmux
> [    1.241153] pinctrl-single 44e10800.pinmux: pin-59 (ocp:P8_30_pinmux) status -22
> [    1.248823] pinctrl-single 44e10800.pinmux: could not request pin 59 (PIN59) from group pinmux_P8_30_default_pin  on device pinctrl-single
> [    1.270959] pinctrl-single 44e10800.pinmux: pin PIN54 already requested by panel; cannot claim for ocp:P8_31_pinmux
> [    1.281656] pinctrl-single 44e10800.pinmux: pin-54 (ocp:P8_31_pinmux) status -22
> [    1.289226] pinctrl-single 44e10800.pinmux: could not request pin 54 (PIN54) from group pinmux_P8_31_default_pin  on device pinctrl-single
> [    1.311239] pinctrl-single 44e10800.pinmux: pin PIN55 already requested by panel; cannot claim for ocp:P8_32_pinmux
> [    1.321929] pinctrl-single 44e10800.pinmux: pin-55 (ocp:P8_32_pinmux) status -22
> [    1.331876] pinctrl-single 44e10800.pinmux: could not request pin 55 (PIN55) from group pinmux_P8_32_default_pin  on device pinctrl-single
> [    1.361067] pinctrl-single 44e10800.pinmux: pin PIN53 already requested by panel; cannot claim for ocp:P8_33_pinmux
> [    1.374309] pinctrl-single 44e10800.pinmux: pin-53 (ocp:P8_33_pinmux) status -22
> [    1.384373] pinctrl-single 44e10800.pinmux: could not request pin 53 (PIN53) from group pinmux_P8_33_default_pin  on device pinctrl-single
> [    1.414032] pinctrl-single 44e10800.pinmux: pin PIN51 already requested by panel; cannot claim for ocp:P8_34_pinmux
> [    1.427251] pinctrl-single 44e10800.pinmux: pin-51 (ocp:P8_34_pinmux) status -22
> [    1.437309] pinctrl-single 44e10800.pinmux: could not request pin 51 (PIN51) from group pinmux_P8_34_default_pin  on device pinctrl-single
> [    1.467049] pinctrl-single 44e10800.pinmux: pin PIN52 already requested by panel; cannot claim for ocp:P8_35_pinmux
> [    1.480296] pinctrl-single 44e10800.pinmux: pin-52 (ocp:P8_35_pinmux) status -22
> [    1.490376] pinctrl-single 44e10800.pinmux: could not request pin 52 (PIN52) from group pinmux_P8_35_default_pin  on device pinctrl-single
[... cut ...]
> Bus 001 Device 005: ID 413c:2002 Dell Computer Corp. SK-8125 Keyboard
> Bus 001 Device 004: ID 413c:1002 Dell Computer Corp. Keyboard Hub
> Bus 001 Device 003: ID 0461:4d22 Primax Electronics, Ltd
> Bus 001 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
> Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
> END

Note: I am still using the original MLO/U-boot provided by GoEmbed as I cannot get it to boot when using the u-boot compiled for "am335x_evm_config" - even after changing the V_OSCK to 25MHz as used by goembed.

Thank you for your help

closed 2019-07-29 by tehkillerbee