beagleboard/linux #209: Cannot enable magnetometer on BeagleBone Blue via kernel driver

jadonk, 2019-07-11:
Please set CONFIG_AK8975=m

Tested with:
* https://gist.github.com/jadonk/99288dd94fb16b6c764c7c81c7735dd0
* https://gist.github.com/jadonk/bcc5f1e65f39406800c3c32cb81290b1

```
debian@beaglebone:/var/lib/cloud9$ dmesg | grep ak
[    0.412403] hw-breakpoint: debug architecture 0x4 unsupported.                                                       
[   39.936608] ak8975 3-000c: mounting matrix not found: using identity...                                              
[   39.936658] ak8975 3-000c: 3-000c supply vdd not found, using dummy regulator                                        
[   39.936818] ak8975 3-000c: Linked as a consumer to regulator.0                                                       
[   39.936842] ak8975 3-000c: 3-000c supply vid not found, using dummy regulator                                        
debian@beaglebone:/var/lib/cloud9$ cat /sys/bus/iio/devices/iio\:device2/in_magn_x_raw 
-24
debian@beaglebone:/var/lib/cloud9$ 
```

RobertCNelson, 2019-07-11:
@jadonk but it is enabled.. maybe something else broken with the in -kernel one?

https://github.com/beagleboard/linux/blob/4.19/arch/arm/configs/bb.org_defconfig#L2163

Regards,

RobertCNelson, 2019-07-11:
For reference, here's what we have on teh blue for mainline (it got removed for roboticcape library..)

```
	mpu9250@68 {
￼			compatible = "invensense,mpu9250";
￼			reg = <0x68>;
￼			interrupt-parent = <&gpio3>;
￼			interrupts = <21 IRQ_TYPE_EDGE_RISING>;
￼			i2c-gate {
￼				#address-cells = <1>;
￼				#size-cells = <0>;
￼				ax8975@c {
￼					compatible = "ak,ak8975";
￼					reg = <0x0c>;
￼				};
￼			};
￼		};
￼	
￼		pressure@76 {
￼			compatible = "bosch,bmp280";
￼			reg = <0x76>;
￼		};
```

jadonk, 2019-07-11:
compatible = "ak,ak8975";
￼
no longer matches. remove "ak" or add the full company name.

jadonk, 2019-07-11:
I'm so confused!

I had these overlays working yesterday before I updated my kernel.

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.50-ti-r22 (voodoo@x2-am57xx-beagle-x15-2gb) (gcc version 8.3.0 (Debian 8.3.0-6)) #1buster SMP PREEMPT Mon Jul 8 22:23:19 UTC 2019
[    0.000000] CPU: ARMv7 Processor [413fc082] revision 2 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: TI AM335x BeagleBone Blue
[    0.000000] Memory policy: Data cache writeback
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: UEFI not found.
[    0.000000] cma: Reserved 48 MiB at 0x9c800000
[    0.000000] On node 0 totalpages: 130560
[    0.000000]   Normal zone: 1148 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 130560 pages, LIFO batch:31
[    0.000000] CPU: All CPU(s) started in SVC mode.
[    0.000000] AM335X ES2.1 (sgx neon)
[    0.000000] random: get_random_bytes called from start_kernel+0xa0/0x508 with crng_init=0
[    0.000000] percpu: Embedded 18 pages/cpu s43084 r8192 d22452 u73728
[    0.000000] pcpu-alloc: s43084 r8192 d22452 u73728 alloc=18*4096
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129412
[    0.000000] Kernel command line: console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 cape_universal=enable quiet
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 437388K/522240K available (13312K kernel code, 1152K rwdata, 4432K rodata, 1024K init, 349K bss, 35700K reserved, 49152K cma-reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xe0000000 - 0xff800000   ( 504 MB)
                   lowmem  : 0xc0000000 - 0xdfe00000   ( 510 MB)
                   pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
                   modules : 0xbf000000 - 0xbfe00000   (  14 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (14304 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   (1024 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   (1153 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 350 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] ftrace: allocating 42466 entries in 125 pages
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=2 to nr_cpu_ids=1.
[    0.000000] 	Tasks RCU enabled.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=1
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] IRQ: Found an INTC at 0x(ptrval) (revision 5.0) with 128 interrupts
[    0.000000] OMAP clockevent source: timer2 at 24000000 Hz
[    0.000020] sched_clock: 32 bits at 24MHz, resolution 41ns, wraps every 89478484971ns
[    0.000039] clocksource: timer1: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 79635851949 ns
[    0.000051] OMAP clocksource: timer1 at 24000000 Hz
[    0.001600] timer_probe: no matching timers found
[    0.001944] Console: colour dummy device 80x30
[    0.001975] WARNING: Your 'console=ttyO0' has been replaced by 'ttyS0'
[    0.001979] This ensures that you still see kernel messages. Please
[    0.001984] update your kernel commandline.
[    0.002044] Calibrating delay loop... 995.32 BogoMIPS (lpj=1990656)
[    0.021473] pid_max: default: 32768 minimum: 301
[    0.021760] Security Framework initialized
[    0.021773] Yama: becoming mindful.
[    0.021922] AppArmor: AppArmor initialized
[    0.022031] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.022043] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.023148] CPU: Testing write buffer coherency: ok
[    0.023212] CPU0: Spectre v2: using BPIALL workaround
[    0.023702] CPU0: thread -1, cpu 0, socket -1, mpidr 0
[    0.045566] Setting up static identity map for 0x80100000 - 0x80100060
[    0.053477] rcu: Hierarchical SRCU implementation.
[    0.064749] EFI services will not be available.
[... 312 lines cut ...]
[   46.034163] wlan0: send auth to 36:c2:13:af:ad:d1 (try 1/3)
[   46.059700] wlan0: authenticated
[   46.068161] wlan0: associate with 36:c2:13:af:ad:d1 (try 1/3)
[   46.077661] wlan0: RX AssocResp from 36:c2:13:af:ad:d1 (capab=0x431 status=0 aid=1)
[   46.111426] wlan0: associated
[   46.127058] wlan0: Limiting TX power to 30 (30 - 0) dBm as advertised by 36:c2:13:af:ad:d1
[   46.489421] cryptd: max_cpu_qlen set to 1000
[   46.594908] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   46.630041] wlcore: Association completed.
[   48.372970] wlcore: down
[   48.524498] IPv6: ADDRCONF(NETDEV_CHANGE): SoftAp0: link becomes ready
[  192.816309] wlcore: Beacon loss detected. roles:0x1
[  196.701489] wlcore: Beacon loss detected. roles:0x1
[  197.626940] wlcore: Connection loss work (role_id: 0).
[  197.626991] wlan0: Connection to AP 36:c2:13:af:ad:d1 lost
[  197.905011] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  198.539945] wlan0: authenticate with 36:c2:13:af:ad:d1
[  198.548613] wlan0: send auth to 36:c2:13:af:ad:d1 (try 1/3)
[  198.672347] wlan0: authenticated
[  198.680061] wlan0: associate with 36:c2:13:af:ad:d1 (try 1/3)
[  198.684771] wlan0: RX AssocResp from 36:c2:13:af:ad:d1 (capab=0x431 status=0 aid=1)
[  198.704419] wlan0: associated
[  198.749922] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  198.775715] wlcore: Association completed.
```

pdp7, 2020-06-09:
@jadonk @RobertCNelson is this still an issue on the Blue?

ConnorRigby, 2022-03-25:
Any news on this one? i'm trying to build a custom kernel that uses this device, but  the latest kernel doesn't seem to even reference the `mpu9250` symbol at all

ConnorRigby, 2022-03-25:
okay, i got it working with:
```
+	mpu9250@68 {
+		compatible = "invensense,mpu9250";
+		reg = <0x68>;
+		interrupt-parent = <&gpio3>;
+		interrupts = <21 IRQ_TYPE_EDGE_RISING>;
+		i2c-gate {
+			#address-cells = <1>;
+			#size-cells = <0>;
+			ax8975@c {
+				compatible = "ak8975";
+				reg = <0x0c>;
+			};
+		};
+	};
```
```
CONFIG_INV_MPU6050_I2C=y
CONFIG_AK8975=y
CONFIG_BMP280=y
```
Hopefully this helps someone in the future