beagleboard/linux #212: Disable SPI0 and enable UART2 on PocketBeagle

josemtrg, 2019-07-16:
I have created an image to a new project using buildroot. Reading the docs it seems that the PocketBeagle has the SPI0 as default, not allowing me to use the UART2 as i need.

What dts file do i need to change to disable SPI0 and enable UART2 by default?
Do i need to change am335x-pocketbeagle.dts or am335x-pocketbeagle-common.dts files?

And if so what should be changed?

I have reached a result but it does not seem to me the right way of doing it...
I have included am335x-bone-pinmux-ttyS2.dtsi on am335x-pocketbeagle.dts and the UART2 started working but on dmesg i have this error:

`[    2.225615] bone-pinmux-helper: probe of ocp:P1_08_pinmux failed with error -22
[    2.233390] pinctrl-single 44e10800.pinmux: pin 44e10954.0 already requested by 48024000.serial; cannot claim for ocp:P1_10_p
inmux
[    2.245239] pinctrl-single 44e10800.pinmux: pin-85 (ocp:P1_10_pinmux) status -22
[    2.252688] pinctrl-single 44e10800.pinmux: could not request pin 85 (44e10954.0) from group pinmux_P1_10_default_pin  on dev
ice pinctrl-single
[    2.265628] bone-pinmux-helper ocp:P1_10_pinmux: Error applying setting, reverse things back
[    2.274547] pinctrl-single 44e10800.pinmux: pin 44e10954.0 already requested by 48024000.serial; cannot claim for ocp:P1_10_p
inmux
[    2.286371] pinctrl-single 44e10800.pinmux: pin-85 (ocp:P1_10_pinmux) status -22
[    2.293818] pinctrl-single 44e10800.pinmux: could not request pin 85 (44e10954.0) from group pinmux_P1_10_default_pin  on dev
ice pinctrl-single
[    2.306757] bone-pinmux-helper ocp:P1_10_pinmux: Error applying setting, reverse things back
[    2.315246] bone-pinmux-helper ocp:P1_10_pinmux: Failed to select default state
`
Now i am not sure if i have correctly disabled the SPI0.

RobertCNelson, 2019-07-16:
@josemtrg , it's up to you, either use config-pin:

https://github.com/beagleboard/bb.org-overlays/blob/master/tools/pmunts_muntsos/config-pin.c

or use this patch:
```
diff --git a/src/arm/am335x-pocketbeagle.dts b/src/arm/am335x-pocketbeagle.dts
index 1932521..95dcc83 100644
--- a/src/arm/am335x-pocketbeagle.dts
+++ b/src/arm/am335x-pocketbeagle.dts
@@ -180,7 +180,7 @@
 
 	/* P1_06 (ZCZ ball A16) spi0_cs0 */
 	P1_06_default_pin: pinmux_P1_06_default_pin { pinctrl-single,pins = <
-		AM33XX_IOPAD(0x095c, PIN_OUTPUT_PULLUP | INPUT_EN | MUX_MODE0) >; };	/* spi0_cs0.spi0_cs0 */
+		AM33XX_IOPAD(0x095c, PIN_OUTPUT | INPUT_EN | MUX_MODE7) >; };		/* spi0_cs0.gpio0_5 */
 	P1_06_gpio_pin: pinmux_P1_06_gpio_pin { pinctrl-single,pins = <
 		AM33XX_IOPAD(0x095c, PIN_OUTPUT | INPUT_EN | MUX_MODE7) >; };		/* spi0_cs0.gpio0_5 */
 	P1_06_gpio_pu_pin: pinmux_P1_06_gpio_pu_pin { pinctrl-single,pins = <
@@ -202,7 +202,7 @@
 
 	/* P1_08 (ZCZ ball A17) spi0_sclk */
 	P1_08_default_pin: pinmux_P1_08_default_pin { pinctrl-single,pins = <
-		AM33XX_IOPAD(0x0950, PIN_OUTPUT_PULLUP | INPUT_EN | MUX_MODE0) >; };	/* spi0_sclk.spi0_sclk */
+		AM33XX_IOPAD(0x0950, PIN_OUTPUT_PULLUP | INPUT_EN | MUX_MODE1) >; };	/* spi0_sclk.uart2_rxd */
 	P1_08_gpio_pin: pinmux_P1_08_gpio_pin { pinctrl-single,pins = <
 		AM33XX_IOPAD(0x0950, PIN_OUTPUT | INPUT_EN | MUX_MODE7) >; };		/* spi0_sclk.gpio0_2 */
 	P1_08_gpio_pu_pin: pinmux_P1_08_gpio_pu_pin { pinctrl-single,pins = <
@@ -226,7 +226,7 @@
 
 	/* P1_10 (ZCZ ball B17) spi0_d0 */
 	P1_10_default_pin: pinmux_P1_10_default_pin { pinctrl-single,pins = <
-		AM33XX_IOPAD(0x0954, PIN_OUTPUT_PULLUP | INPUT_EN | MUX_MODE0) >; };	/* spi0_d0.spi0_d0 */
+		AM33XX_IOPAD(0x0954, PIN_OUTPUT_PULLUP | INPUT_EN | MUX_MODE1) >; };	/* spi0_d0.uart2_txd */
 	P1_10_gpio_pin: pinmux_P1_10_gpio_pin { pinctrl-single,pins = <
 		AM33XX_IOPAD(0x0954, PIN_OUTPUT | INPUT_EN | MUX_MODE7) >; };		/* spi0_d0.gpio0_3 */
 	P1_10_gpio_pu_pin: pinmux_P1_10_gpio_pu_pin { pinctrl-single,pins = <
@@ -250,7 +250,7 @@
 
 	/* P1_12 (ZCZ ball B16) spi0_d1 */
 	P1_12_default_pin: pinmux_P1_12_default_pin { pinctrl-single,pins = <
-		AM33XX_IOPAD(0x0958, PIN_OUTPUT_PULLUP | INPUT_EN | MUX_MODE0) >; };	/* spi0_d1.spi0_d1 */
+		AM33XX_IOPAD(0x0958, PIN_OUTPUT | INPUT_EN | MUX_MODE7) >; };		/* spi0_d1.gpio0_4 */
 	P1_12_gpio_pin: pinmux_P1_12_gpio_pin { pinctrl-single,pins = <
 		AM33XX_IOPAD(0x0958, PIN_OUTPUT | INPUT_EN | MUX_MODE7) >; };		/* spi0_d1.gpio0_4 */
 	P1_12_gpio_pu_pin: pinmux_P1_12_gpio_pu_pin { pinctrl-single,pins = <
```

josemtrg, 2019-07-16:
@RobertCNelson i was not able to apply the patch because it was complaining that it could not found the file to apply the patch.

So what i did was the following:

1. Created a clean image.
2. Then i went to /output/build/linux-4.4/arch/arm/boot/dts/am335x-pocketbeagle-common.dtsi
3. Made the changes you proposed.
4. make linux-rebuild
5. Installed the image and everything is working as i wanted.

@RobertCNelson, Thank you for your time!

closed 2019-07-16 by josemtrg