beagleboard/linux #213: LED netdev trigger

jadonk, 2019-07-25:
Please turn on the NETDEV and all the fun LED triggers. :-)

This is needed for BeagleBone AI WiFi link indication.

pdp7, 2020-06-09:
@jadonk @RobertCNelson did we do this?

RobertCNelson, 2020-06-09:
Yes

pdp7, 2020-06-09:
@RobertCNelson 

`drivers/leds/trigger/ledtrig-netdev.c` added in 4.14:
```
5f820ed52371 v5.5-rc1~98^2      leds: trigger: netdev: fix handling on interface rename
909346433064 v5.1-rc3~9^2       leds: trigger: netdev: use memcpy in device_name_store
4cb6560514fa v5.1-rc3~9^2~2     leds: trigger: netdev: fix refcnt leak on interface rename
f8112a1de1a7 v4.19-rc1~161^2~10 leds: netdev trigger: simplifications from core changes
2282e125a406 v4.19-rc1~161^2~15 leds: triggers: let struct led_trigger::activate() return an error code
06f502f57d0d v4.16-rc1~136^2~11 leds: trigger: Introduce a NETDEV trigger
```

If it is 4.19 or 5.4, then I think we just need to add:
```
CONFIG_LEDS_TRIGGER_NETDEV=m
```

RobertCNelson, 2020-06-09:
it's been enabled, and shipped with teh bbai out of box image.

pdp7, 2020-06-09:
@RobertCNelson  Ah, I was editing while you were writing.  What did we add to the device tree for it?

RobertCNelson, 2020-06-09:
https://github.com/beagleboard/BeagleBoard-DeviceTrees/commit/37d95fa2d8797613eb69ba64c36c73250fb6e9dc#diff-5e2ddc241fee7ebdc0dee927b7061de1

pdp7, 2020-06-09:
Ah I see:
https://github.com/beagleboard/BeagleBoard-DeviceTrees/blob/v4.19.x-ti/src/arm/am5729-beagleboneai.dts
```
		led4 {
			label = "beaglebone:green:usr4";
			gpios = <&gpio3 7 GPIO_ACTIVE_HIGH>;
			linux,default-trigger = "netdev";
			default-state = "off";
		};
```

pdp7, 2020-06-09:
Thanks,  I am out of phase :)

So close it?

closed 2020-06-09 by RobertCNelson