beagleboard/linux #214: Enabling Pocketbeagle SPI1 for ENC28J60

josemtrg, 2019-07-26:
Hello there! I have developed my own OS to a Pocketbeagle based on buildroot.

Basically i need to use UART0, 2 and 4, USB1 and SPI1 to communicate with an ENC28J60.

This is what i have done so far:

Created my own dts file like this:
[am335x-unprint-pocketbeagle.txt](https://github.com/beagleboard/linux/files/3435390/am335x-unprint-pocketbeagle.txt)

And rebuild everything! I have tested the UARTs and USB and everything is working. So far so good...

Now i am having problems setting up the SPI1 from my tests to each one of the pins MOSI,MISO, CLK or CS they never change state. I have set up the SPI and ENC28J60 based on 
PB-SPI1-ETH-CLICK.dts and only changed INT and RST pins.

On the PB-SPI1-ETH-CLICK.dts file i can see that the INT and RESET pin are set up as inputs!?
For me, it does not make sense since the RESET pin is an output and INT an input indeed.

So i changed in what i think is the correct way:

AM33XX_IOPAD(0x0998, PIN_OUTPUT_PULLUP | INPUT_EN | MUX_MODE7)/*mcasp0_axr0.gpio3_16 RST */
AM33XX_IOPAD(0x09a4, PIN_INPUT | MUX_MODE7) /* mcasp0_fsr.gpio3_19 INT */ 

Regarding the SPI pins was well since that on that file every pin is set up as input. But CLK and MOSI should be set up as outputs

After boot, it says that the enc28j60 driver has started and working, and also tries to acquire an IP address through DHCP but it cant.

The ENC28J60 is connected to a RJ45 connector with LEDs but it does work or shows some LED blinking.

Am I forgetting something? Can someone help me?

By the way im using kernel 4.4.155

Thank you for your time!

pdp7, 2020-06-09:
@josemtrg are you still having this problem?

josemtrg, 2020-06-09:
> @josemtrg are you still having this problem?

@pdp7 I have already solved it.

closed 2020-06-09 by pdp7