beagleboard/linux #215: BBB UART sometimes skips first byte and duplicates last one

spanceac, 2019-08-08:
Hi,

We observed some erratic UART behavior on BBB using this kernel at tag `4.14.53-ti-r62`.

Fortunately, this is very easy to reproduce.

Take a BBB and load the UART4 overlay:
```
enable_uboot_overlays=1
uboot_overlay_addr0=/boot/overlays/BB-UART4-00A0.dtbo
```

Connect the RX of a serial cable to the UART4 TX(pin P9_13) and open a terminal on this serial cable with 9600 8N1 settings.

On the BBB, run the following script that echoes some text in a loop on UART4:
```
while true; do
echo 123456789abcdefghijklmnopqrstuvxyz > /dev/ttyS4
done
```

The expected output is:
```
123456789abcdefghijklmnopqrstuvxyz
123456789abcdefghijklmnopqrstuvxyz
123456789abcdefghijklmnopqrstuvxyz
123456789abcdefghijklmnopqrstuvxyz
```

But after some time(usually less than 60s) you will see an output of this type:
```
123456789abcdefghijklmnopqrstuvxyz
123456789abcdefghijklmnopqrstuvxyz
23456789abcdefghijklmnopqrstuvxyz

123456789abcdefghijklmnopqrstuvxyz
123456789abcdefghijklmnopqrstuvxyz
```

You can notice how the first byte `1` disappeared and the last byte(new line) was duplicated.

Sometimes it happens that only the first byte is skipped but the last one is not duplicated, output becoming:
```
123456789abcdefghijklmnopqrstuvxyz
23456789abcdefghijklmnopqrstuvxyz
123456789abcdefghijklmnopqrstuvxyz
123456789abcdefghijklmnopqrstuvxyz
```

Do you know what could cause this and if there is an existing fix?

Thank you!

spanceac, 2019-08-09:
I also want to mention that this problem is specific to the OMAP 8250 driver. 

The other OMAP serial driver doesn't exhibit this problem.

The problem reproduces if we activate `CONFIG_SERIAL_8250_OMAP` in the kernel.

If instead we activate `CONFIG_SERIAL_OMAP` in the kernel, the problem will not reproduce.

pdp7, 2020-06-09:
@spanceac @RobertCNelson is there a reason that `CONFIG_SERIAL_OMAP` can not just be used?

pdp7, 2020-06-09:
@RobertCNelson so I should test this as we only support CONFIG_SERIAL_8250_OMAP in the current kernel builds, right?