beagleboard/linux #220: Q: How is the bb.org_defconfig maintained?

fgervais, 2019-10-24:
I'm using the bb.org_defconfig as a base for a project similar to the beagleboard but I'm aiming for a minimal config, specific to our needs.

What I noticed is that bb.org_defconfig makes a really good job at specifying only the options that differs from the defaults. Also even if it's a file containing cherry-picked configs, it maintains the proper order of configs.

Is it all done manually or there is a special trick to achieve this?

Is the person maintaining this file manually figuring out if a config is default or not and manually inserting it at the correct position? I know this is what I'm doing right now, that's is why I am curious if there is something better :).

RobertCNelson, 2019-10-24:
@fgervais , i cheat and use savedefconfig, right before it's pushed..

```
cp ${DIR}/patches/defconfig ${DIR}/KERNEL/.config
make ARCH=${KERNEL_ARCH} savedefconfig
cp ${DIR}/KERNEL/defconfig ${DIR}/KERNEL/arch/${KERNEL_ARCH}/configs/${example}_defconfig
${git_bin} add arch/${KERNEL_ARCH}/configs/${example}_defconfig
```

Regards,

fgervais, 2019-10-24:
Well I'm glad I asked. I didn't know about this `savedefconfig` even though it seems it's been there for a while. Thanks @RobertCNelson.

closed 2019-10-24 by fgervais