beagleboard/linux #223: PocketBeagle devicetree conflict(?) between P1_31 and mmc0_sdwp

ikarso, 2020-01-14:
According to the SRM & schematics the P1_31 are connected to PRU0.4 / ZCZ B12.
But in the device tree for Linux kernel 5.4 the MMC0 has also a reference to the ZCZ B12 pin
Line 91:   AM33XX_PADCONF(AM335X_PIN_MCASP0_ACLKR, PIN_INPUT, MUX_MODE4)		/* (B12) mcasp0_aclkr.mmc0_sdwp */

https://github.com/beagleboard/linux/blob/5.4/arch/arm/boot/dts/am335x-pocketbeagle.dts

Is it any point to have the SD Write protect exposed on the header?
On the Techlab cape the P1_31 are connected to MMA8453Q interrupt line2 and will be a little pain in the accelerator overlay to free the pin on the mmc0.

RobertCNelson, 2020-01-14:
@ikarso, Yeah we rip it out:

https://github.com/beagleboard/BeagleBoard-DeviceTrees/commit/0ecfcfdfba3268708a0293a86b04bcd397cf1530

If you want to submit it mainline go ahead, i got behind on things so i didn't..

Regards,

pdp7, 2020-06-09:
@ikarso @RobertCNelson I see it is still there in mainline:
https://elixir.bootlin.com/linux/latest/source/arch/arm/boot/dts/am335x-pocketbeagle.dts#L91

I'll submit https://github.com/beagleboard/BeagleBoard-DeviceTrees/commit/0ecfcfdfba3268708a0293a86b04bcd397cf1530 to the devicetree mailing list

pdp7, 2020-06-09:
I've posted the patch:
https://lore.kernel.org/linux-omap/20200609214521.GB2995279@x1/

closed 2020-06-09 by pdp7