beagleboard/linux #224: Beaglebone Black HDMI

kzoltaan, 2020-01-19:
I give up. I unable to bring up the HDMI on bbb. Cable is okay. Works with other old bbb... Monitor Samsung S24F350 and S22F350. 5V GND and hdmi shield connected together. No console on HDMI, and no GUI...

I tried with bone-debian-9.9-lxqt-armhf-2019-08-03-4gb.
Here are the logs:
boot:
https://pastebin.com/g0P7W60D

cat lightdm.log
https://pastebin.com/cvvU9Hpm

cat x-0.log
https://pastebin.com/KJjCfB0b

cat Xorg.0.log
https://pastebin.com/bSmMaNb5

xrandr: takes Failed to get size of gamma for output default message
`debian@beaglebone:/var/log$ export DISPLAY=:0
debian@beaglebone:/var/log$ xrandr --output default --mode 1280x800
xrandr: Failed to get size of gamma for output default
debian@beaglebone:/var/log$
`
`debian@beaglebone:/var/log$ xrandr --listmonitors
xrandr: Failed to get size of gamma for output default
Monitors: 1
 0: +default 1280/339x800/212+0+0  default
debian@beaglebone:/var/log$
`
parse-edid hangs... crtl-c required....
`debian@beaglebone:/sys/class/drm/card1-HDMI-A-1$ ls -la
total 0
drwxr-xr-x 3 root root    0 Jan 19 19:52 .
drwxr-xr-x 4 root root    0 Jan 19 19:52 ..
lrwxrwxrwx 1 root root    0 Jan 19 19:52 device -> ../../card1
-r--r--r-- 1 root root 4096 Jan 19 19:52 dpms
-r--r--r-- 1 root root    0 Jan 19 19:52 edid
-r--r--r-- 1 root root 4096 Jan 19 19:52 enabled
-r--r--r-- 1 root root 4096 Jan 19 19:52 modes
drwxr-xr-x 2 root root    0 Jan 19 19:52 power
-rw-r--r-- 1 root root 4096 Jan 19 19:52 status
lrwxrwxrwx 1 root root    0 Jan 19 19:52 subsystem -> ../../../../../../../class/drm
-rw-r--r-- 1 root root 4096 Jan 19 19:52 uevent
debian@beaglebone:/sys/class/drm/card1-HDMI-A-1$  parse-edid edid
^C
debian@beaglebone:/sys/class/drm/card1-HDMI-A-1$
`
 
`root@beaglebone:/opt/scripts/tools# ./version.sh
git:/opt/scripts/:[109f74fb87e6034ae1a8971a244064a8d5e090a5]
eeprom:[A335BNLT00C03919BBBK26EB]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2019-08-03]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-gbb4af0f50f]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2019.04-00002-g3d8c979660]:[location: dd MBR]
kernel:[4.14.108-ti-r113]
nodejs:[v6.17.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190801.0-0rcnee0~stretch+20190801]
pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~stretch+20190227]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.4-git20190227.1-0rcnee0~stretch+20190327]
pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev gpio pwm eqep remoteproc admin spi tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 video=HDMI-A-1:1280x800@60e]
dmesg | grep remote
[    2.712676] remoteproc remoteproc0: wkup_m3 is available
[    3.270376] remoteproc remoteproc0: powering up wkup_m3
[    3.279816] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    3.300123] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
dmesg | grep pinctrl-single
[    0.942152] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    0.954192] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
`
Whats wrong?
Thanks

kzoltaan, 2020-01-20:
HDMI in Debian 7.9 2015-11-12 4GB SD LXDE image is working out of the box.

kzoltaan, 2020-01-20:
Is any hardware difference between BBB production series?

RobertCNelson, 2020-01-20:
@kzoltaan your comparing a very hacked up 3.8.x based kernel in the old 2015 image vs mainline.  I'd say to retry with v4.19.x (but it's not as good as v4.14.x) and i don't really have v5.4.x in a shape for testing..

So yes, the old hacked up v3.8.x's video did work better for some monitors/resolutions..

Sorry, i'm in no position to analyse the hdmi signal or to personally pick-up that monitor..

Regards,

kzoltaan, 2020-01-20:
I have 6 pcs brand new bbb (4.14.71-ti-r80). I want to use the first one. Not working. I played with this when post borned. The second one is working, I made an SD card from eMMC. Third BBB working... I have 2 pcs old bbb. First old bbb HDMI not working with this image(worked with old linux). While I flashing other old one, I got these messages, and sometimes HDMI blinking
<pre>
[  566.537644] tilcdc 4830e000.lcdc: tilcdc_crtc_irq(0x00000004): Sync lost
[  566.853885] tilcdc 4830e000.lcdc: tilcdc_crtc_irq(0x00000020): FIFO underflow
[  575.612879] tilcdc 4830e000.lcdc: tilcdc_crtc_irq(0x00000020): FIFO underflow
[  575.627779] tilcdc_crtc_irq: 40 callbacks suppressed
[  575.627803] tilcdc 4830e000.lcdc: tilcdc_crtc_irq(0x00000004): Sync lost
[  575.639615] tilcdc 4830e000.lcdc: tilcdc_crtc_irq(0x00000004): Sync lost
[  575.646392] tilcdc 4830e000.lcdc: tilcdc_crtc_irq(0x00000104): Sync lost
[  575.661135] tilcdc 4830e000.lcdc: tilcdc_crtc_irq(0x00000004): Sync lost
[  575.668003] tilcdc 4830e000.lcdc: tilcdc_crtc_irq(0x00000004): Sync lost
[  575.677779] tilcdc 4830e000.lcdc: tilcdc_crtc_irq(0x00000004): Sync lost
[  575.684660] tilcdc 4830e000.lcdc: tilcdc_crtc_irq(0x00000004): Sync lost
[  575.694429] tilcdc 4830e000.lcdc: tilcdc_crtc_irq(0x00000004): Sync lost
[  575.701302] tilcdc 4830e000.lcdc: tilcdc_crtc_irq(0x00000004): Sync lost
[  575.711101] tilcdc 4830e000.lcdc: tilcdc_crtc_irq(0x00000004): Sync lost
[  576.060982] tilcdc 4830e000.lcdc: tilcdc_crtc_irq(0x00000004): Sync lost flood detected, recovering
[  576.070292] tilcdc 4830e000.lcdc: tilcdc_crtc_recover_work: Reset CRTC
</pre>

But this is working with this image.... no HDMI blinks... I dont understand... Possible scrap brand new products.... ?

RobertCNelson, 2020-01-20:
@kzoltaan the "FIFO underflow" should be fixed in u-boot, it was a ddr timing issue.

https://github.com/u-boot/u-boot/commit/8c17cbdf8a8023abdd0009af4dc9dbc0541b4a0f

This can still happen if the version of u-boot on the eMMC is older then v2019.04, when you run ./version.sh

Regards,

pdp7, 2020-06-16:
Re-open if still issue

closed 2020-06-16 by pdp7

kzoltaan, 2020-11-27:
I picked up again this problem, but still not working. 
Here is version sh:
<pre>
`root@beaglebone:/opt/scripts/tools# ./version.sh
git:/opt/scripts/:[1aa73453b2c980b75e31e83dab7dd8b6696f10c7]
eeprom:[A335BNLT00C04219BBBK0683]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-10-07]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.09-00002-g0b54a51eee]:[location: dd MBR]
kernel:[4.14.71-ti-r80]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[enable_uboot_cape_universal=0]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20180928.0-0rcnee0~stretch+20180928]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.3-git20181005.0-0rcnee0~stretch+20181005]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet video=HDMI-A-1:1280x800@60]
dmesg | grep pinctrl-single
[    1.133218] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.145397] gpio-of-helper ocp:cape-universal: ready
END
`
</pre>
I see the device:
<pre>
root@beaglebone:/sys/class/drm/card1-HDMI-A-1# ls
device  dpms  edid  enabled  modes  power  status  subsystem  uevent
root@beaglebone:/sys/class/drm/card1-HDMI-A-1# cat status
connected
root@beaglebone:/sys/class/drm/card1-HDMI-A-1# cat modes
1280x800
1280x720
1280x720
1280x720
1024x768
1024x768
1024x768
832x624
800x600
800x600
800x600
800x600
720x576
720x480
720x480
640x480
640x480
640x480
640x480
640x480
720x400

</pre>
edid shows the type of monitor... (I cant install parse-edid now...)

but monitor shows nothing....

RobertCNelson, 2020-11-27:
```
dogtag:[BeagleBoard.org Debian Image 2018-10-07]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.09-00002-g0b54a51eee]:[location: dd MBR]
kernel:[4.14.71-ti-r80]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[enable_uboot_cape_universal=0]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade ]
pkg:[bb-cape-overlays]:[4.4.20180928.0-0rcnee0~stretch+20180928]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
```

This is such a random combination..

Please test with a spare microSD card.

https://rcn-ee.net/rootfs/bb.org/testing/2020-11-23/buster-lxqt/bone-debian-10.6-lxqt-armhf-2020-11-23-4gb.img.xz

and remember to hold the "user" button next to the microsd/usb before plugging in power..

and then add ```sudo /opt/scripts/tools/version.sh```

Regards,

kzoltaan, 2020-11-30:
not working.
I have a board which is working with my version.
We are using 5-6-7 pcs bbb for operator display. One of them worked few month then stopped. No HDMI output. restart, HDMI working, restart again HDMI not working. 
Some bbb not working out of the box... Some working... Motitors and cables are same.... No difference between the systems. I made a master SD card, and use emmc-flasher script....  Is that possible that the brand new bbb is a scrap?  i'm not think so
<pre>

`root@beaglebone:/sys/class/drm/card0/card0-HDMI-A-1# /opt/scripts/tools/version.sh
git:/opt/scripts/:[20c080bc91f243ef852a704758788a31083b0f9e]
eeprom:[A335BNLT00C04219BBBK069B]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Buster LXQt Image 2020-11-23]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-gc9b3922522]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.09-00002-g0b54a51eee]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot-univ.dts]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
UBOOT: Loaded Overlay:[BB-HDMI-TDA998x-00A0]
kernel:[4.19.94-ti-r57]
nodejs:[v10.21.0]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20201119.0-0~buster+20201123]
pkg:[bb-customizations]:[1.20201105.0-0~buster+20201110]
pkg:[bb-usb-gadgets]:[1.20200504.0-0~buster+20200504]
pkg:[bb-wl18xx-firmware]:[1.20200813.1-0~buster+20200813]
pkg:[kmod]:[26-1]
pkg:[librobotcontrol]:[1.0.5-git20200715.0-0~buster+20200716]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal input bluetooth netdev i2c gpio admin spi iio docker tisdk weston-launch xenomai cloud9ide pwm eqep remoteproc]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet video=HDMI-A-1:1280x720@60e]
dmesg | grep remote
[  102.504494] remoteproc remoteproc0: wkup_m3 is available
[  102.515962] remoteproc remoteproc0: powering up wkup_m3
[  102.515996] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[  102.516279] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
dmesg | grep pinctrl-single
[    1.177948] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    1.191331] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
`

`root@beaglebone:/sys/class/drm/card0/card0-HDMI-A-1# edid-decode edid
EDID version: 1.3
Manufacturer: SAM Model d1a Serial Number 1515867222
Made in week 36 of 2019
Digital display
Maximum image size: 48 cm x 27 cm
Gamma: 2.20
DPMS levels: Off
RGB color display
First detailed timing is preferred timing
Display x,y Chromaticity:
  Red:   0.6455, 0.3369
  Green: 0.3281, 0.6152
  Blue:  0.1464, 0.0556
  White: 0.3134, 0.3291
Established timings supported:
  720x400@70Hz 9:5 HorFreq: 31469 Hz Clock: 28.320 MHz
  640x480@60Hz 4:3 HorFreq: 31469 Hz Clock: 25.175 MHz
  640x480@67Hz 4:3 HorFreq: 35000 Hz Clock: 30.240 MHz
  640x480@72Hz 4:3 HorFreq: 37900 Hz Clock: 31.500 MHz
  640x480@75Hz 4:3 HorFreq: 37500 Hz Clock: 31.500 MHz
  800x600@56Hz 4:3 HorFreq: 35200 Hz Clock: 36.000 MHz
  800x600@60Hz 4:3 HorFreq: 37900 Hz Clock: 40.000 MHz
[... 48 lines cut ...]
Detailed mode: Clock 27.000 MHz, 477 mm x 268 mm
                720  732  796  864 hborder 0
                576  581  586  625 vborder 0
               -hsync -vsync
               VertFreq: 50 Hz, HorFreq: 31250 Hz
Detailed mode: Clock 27.000 MHz, 477 mm x 268 mm
                720  736  798  858 hborder 0
                480  489  495  525 vborder 0
               -hsync -vsync
               VertFreq: 59 Hz, HorFreq: 31468 Hz
Checksum: 0x50 (valid)
`
</pre>
Your linux drop now like this:

<pre>
` 102.516279] remoteproc remoteproc0: remote processor wkup_m3 is now up
[  102.516308] wkup_m3_ipc 44e11324.wkup_m3_ipc: CM3 Firmware Version = 0x193
[  102.614630] PM: bootloader does not support rtc-only!
[  278.791677] sysrq: SysRq : HELP : loglevel(0-9) reboot(b) crash(c) terminate-all-tasks(e) memory-full-oom-kill(f) kill-all-tasks(i) thaw-filesystems(j) sak(k) show-backtrace-all-active-cpus(l) show-memory-usage(m) nice-all-RT-tasks(n) poweroff(o) show-registers(p) show-all-timers(q) unraw(r) sync(s) show-task-states(t) unmount(u) force-fb(V) show-blocked-tasks(w) dump-ftrace-buffer(z)
[  278.826049] sysrq: SysRq : HELP : loglevel(0-9) reboot(b) crash(c) terminate-all-tasks(e) memory-full-oom-kill(f) kill-all-tasks(i) thaw-filesystems(j) sak(k) show-backtrace-all-active-cpus(l) show-memory-usage(m) nice-all-RT-tasks(n) poweroff(o) show-registers(p) show-all-timers(q) unraw(r) sync(s) show-task-states(t) unmount(u) force-fb(V) show-blocked-tasks(w) dump-ftrace-buffer(z)
[  279.863919] sysrq: SysRq : HELP : loglevel(0-9) reboot(b) crash(c) terminate-all-tasks(e) memory-full-oom-kill(f) kill-all-tasks(i) thaw-filesystems(j) sak(k) show-backtrace-all-active-cpus(l) show-memory-usage(m) nice-all-RT-tasks(n) poweroff(o) show-registers(p) show-all-timers(q) unraw(r) sync(s) show-task-states(t) unmount(u) force-fb(V) show-blocked-tasks(w) dump-ftrace-buffer(z)
`
</pre>
thanks....

kzoltaan, 2020-11-30:
I found a version, this is working. I try to clone....
<pre>
root@beaglebone:/opt/scripts/tools# ./version.sh
git:/opt/scripts/:[1aa73453b2c980b75e31e83dab7dd8b6696f10c7]
eeprom:[A335BNLT00C03919BBBK2962]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-10-07]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.09-00002-g0b54a51eee]:[location: dd MBR]
kernel:[4.14.71-ti-r80]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20180928.0-0rcnee0~stretch+20180928]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.3-git20181005.0-0rcnee0~stretch+20181005]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet video=HDMI-A-1:1280x800@60e]
dmesg | grep pinctrl-single
[    1.133034] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.145195] gpio-of-helper ocp:cape-universal: ready
END

</pre>

kzoltaan, 2022-02-23:
HI!
I picked up again this project. Agian:  We make terminals for workers with our custom business logic. So we have 7 pcs Samsung S22F350 and we have approx 10 pcs bbb.   4 workspaces are working. We are using same version from bbb, monitors, power supplies and HDMI cables. The working items goes with video=HDMI-A-1:1280x800@60e  I made a new installer for beaglebone. 

version.sh
`git:/opt/scripts/:[a335abcf87d2ef5fd96e7de83cdf3f0ff5a4da2b]
eeprom:[A335BNLT00C04219BBBK06A9]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Bullseye Console Image 2022-02-23]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot SPL 2019.04-00002-gc9b3922522 (Aug 24 2020 - 16:42:18 -0500)]:[location: dd MBR]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-gc9b3922522]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot SPL 2018.09-00002-g0b54a51eee (Sep 10 2018 - 19:41:39 -0500)]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.09-00002-g0b54a51eee]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-UIO-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0.kernel]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0.kernel]
UBOOT: Loaded Overlay:[BB-HDMI-TDA998x-00A0.kernel]
kernel:[5.10.100-ti-r37]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=AM335X-PRU-UIO-00A0.dtbo]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20210821.0-0~bullseye+20210821]
pkg:[bb-customizations]:[1.20220119.0-0~bullseye+20220125]
pkg:[bb-usb-gadgets]:[1.20220112.3-0~bullseye+20220112]
pkg:[bb-wl18xx-firmware]:[1.20211222.2-0~bullseye+20211222]
pkg:[kmod]:[28-1]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
pkg:[firmware-ti-connectivity]:[20210315-3]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal input bluetooth netdev i2c gpio admin tisdk weston-launch cloud9ide]
cmdline:[console=ttyS0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet video=HDMI-A-1:1280x720@50e]
dmesg | grep remote
[   34.712487] remoteproc remoteproc0: wkup_m3 is available
[   48.277526] remoteproc remoteproc0: powering up wkup_m3
[   48.277564] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217148
[   48.277854] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
[    0.123597] platform 4a300000.pruss: Cannot lookup hwmod 'pruss'
[    4.229799] sysfs: cannot create duplicate filename '/bus/platform/devices/4a300000.pruss'
dmesg | grep pinctrl-single
[    3.727569] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
`
edid-decode

`root@BeagleBone:/sys/class/drm/card0-HDMI-A-1# edid-decode edid
edid-decode (hex):

00 ff ff ff ff ff ff 00 4c 2d 1a 0d 38 4a 58 43
25 1f 01 03 80 30 1b 78 2a 52 95 a5 56 54 9d 25
0e 50 54 bf ef 80 71 4f 81 c0 81 00 81 80 95 00
a9 c0 b3 00 01 01 02 3a 80 18 71 38 2d 40 58 2c
45 00 dd 0c 11 00 00 1e 00 00 00 fd 00 32 4b 1e
51 11 00 0a 20 20 20 20 20 20 00 00 00 fc 00 53
32 32 46 33 35 30 0a 20 20 20 20 20 00 00 00 ff
00 48 34 4c 52 39 30 30 30 37 33 0a 20 20 01 25

02 03 11 b1 46 90 04 1f 13 12 03 65 03 0c 00 10
00 01 1d 00 bc 52 d0 1e 20 b8 28 55 40 dd 0c 11
00 00 1e 8c 0a d0 90 20 40 31 20 0c 40 55 00 dd
0c 11 00 00 18 8c 0a d0 8a 20 e0 2d 10 10 3e 96
00 dd 0c 11 00 00 18 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 50

----------------
[... 63 lines cut ...]
  Video Data Block:
    VIC  16:  1920x1080   60.000 Hz  16:9    67.500 kHz 148.500 MHz (native)
    VIC   4:  1280x720    60.000 Hz  16:9    45.000 kHz  74.250 MHz
    VIC  31:  1920x1080   50.000 Hz  16:9    56.250 kHz 148.500 MHz
    VIC  19:  1280x720    50.000 Hz  16:9    37.500 kHz  74.250 MHz
    VIC  18:   720x576    50.000 Hz  16:9    31.250 kHz  27.000 MHz
    VIC   3:   720x480    59.940 Hz  16:9    31.469 kHz  27.000 MHz
  Vendor-Specific Data Block (HDMI), OUI 00-0C-03:
    Source physical address: 1.0.0.0
  Detailed Timing Descriptors:
    DTD 2:  1280x720    50.000 Hz  16:9    37.500 kHz  74.250 MHz (477 mm x 268 mm)
                 Hfront  440 Hsync  40 Hback 220 Hpol P
                 Vfront    5 Vsync   5 Vback  20 Vpol P
    DTD 3:   720x576    50.000 Hz   5:4    31.250 kHz  27.000 MHz (477 mm x 268 mm)
                 Hfront   12 Hsync  64 Hback  68 Hpol N
                 Vfront    5 Vsync   5 Vback  39 Vpol N
    DTD 4:   720x480    59.940 Hz   3:2    31.469 kHz  27.000 MHz (477 mm x 268 mm)
                 Hfront   16 Hsync  62 Hback  60 Hpol N
                 Vfront    9 Vsync   6 Vback  30 Vpol N
Checksum: 0x50
`

HDMI is blank. No signal... Do you have any idea how can I solve this? Change monitors? What kind of monitors working with bbb? 

Thanks

RobertCNelson, 2022-02-24:
@kzoltaan i don't think this will fully fix the issue... A few things for display (ddr setting) are done in u-boot..

Please test with:

```
sudo apt update
sudo apt install bb-u-boot-am335x-evm
sudo /opt/u-boot/bb-u-boot-am335x-evm/install.sh
sudo reboot
```

i see you have the override:
```
video=HDMI-A-1:1280x720@50e
```

I'd fully expect that work..

Sorry i don't have a S22F350 to help verify...  by any chance do those monitors have a built in usb hub or usb port?  I've seen with some finicky monitors a bad ground path over the hdmi cables..  For them i just plug a usb cable to connect both..

Regards,