beagleboard/linux #226: Not loading DP83867 Driver

willis2019, 2020-02-26:
I have a development board that is using the DP83867IR IC for Ethernet and spent hours trying to troubleshoot my device tree. The MDIO would kick back GENERIC PHY in "dmesg" when using Bone Debian 9.9 (Kernel 4.19) I went back to an earlier version of the OS (Bone Debian 9.1, Kernel 4.9) and the PHY driver comes back as DP83867 with proper reg. I verfied this twice using new installs (Etcher) of both 9.1 and 9.9. 
In 9.1, I can see the driver module is loaded via "lsmod" and is attached to the PHY. Not sure why 9.9 is not loading the correct Ethernet Driver and I'm not sure this is isolated to just the DP83867 IC. 
Without Device Tree configurations in "dmesg" I can see the driver is substantiated but can't attached to a PHY unless I compile my own overlay with DP83867 node configurations.
If anymore data is needed, please let me now. I can can copy the var/logs files of both kernels to prove what I'm seeing.

RobertCNelson, 2020-02-26:
@willis2019 before I start looking at this, please add the state exact kernel versions referenced above.. (uname -r) as we don't have a lookup list.

Regards,

arun-prathap, 2020-12-29:
> I have a development board that is using the DP83867IR IC for Ethernet and spent hours trying to troubleshoot my device tree. The PHY would kick back GENERIC PHY in "dmesg" when using Bone Debian 9.9 (Kernel 4.19) I went back to an earlier version of the OS (Bone Debian 9.1, Kernel 4.9) and the PHY driver comes back as DP83867 with proper reg. I verified this twice using new installs (Etcher) of both 9.1 and 9.9.
> In 9.1, I can see the driver module is loaded via "lsmod" and is attached to the PHY. Not sure why 9.9 is not loading the correct Ethernet Driver and I'm not sure this is isolated to just the DP83867 IC.
> Without Device Tree configurations, the status in "dmesg" I can see the driver is substantiated but can't attached to a PHY unless I compile my own overlay with DP83867 node configurations.
> If anymore data is needed, please let me now. I can can copy the var/logs files of both kernels to prove what I'm seeing.

Would you like to share how you solved the issue. I am seeing similar issues between BBB kernel (4.14.71-ti-rt80 Generic PHY) and zImage from TI SDK(4.14.40 DP83867 PHY).