beagleboard/linux #231: Image too large: increase CONFIG_SYS_BOOTM_LEN

meghaviparikh, 2020-04-27:
Hi, 
I have been trying to compile the image for BBB using verified u-boot. I am facing the following error. 

```
   Booting using the fdt blob at 0x82aae18c
   Uncompressing Kernel Image
Image too large: increase CONFIG_SYS_BOOTM_LEN: 0x1000000
Required Size minimum: 0x1000000
Must RESET board to recover
resetting ...
```

Here is the full log:

```
U-Boot SPL 2013.04-dirty (Jul 10 2013 - 14:02:53)
musb-hdrc: ConfigData=0xde (UTMI-8, dyn FIFOs, HB-ISO Rx, HB-ISO Tx, SoftConn)
musb-hdrc: MHDRC RTL version 2.0 
musb-hdrc: setup fifo_mode 4
musb-hdrc: 28/31 max ep, 16384/16384 memory
USB Peripheral mode controller at 47401000 using PIO, IRQ 0
musb-hdrc: ConfigData=0xde (UTMI-8, dyn FIFOs, HB-ISO Rx, HB-ISO Tx, SoftConn)
musb-hdrc: MHDRC RTL version 2.0 
musb-hdrc: setup fifo_mode 4
musb-hdrc: 28/31 max ep, 16384/16384 memory
USB Host mode controller at 47401800 using PIO, IRQ 0
OMAP SD/MMC: 0
reading u-boot.img
reading u-boot.img

U-Boot 2020.04-00684-gd202f67db0-dirty (Apr 27 2020 - 18:54:58 -0400)

CPU  : AM335X-GP rev 2.1
Model: TI AM335x BeagleBone Black
DRAM:  512 MiB
MMC:   OMAP SD/MMC: 0, OMAP SD/MMC: 1
Loading Environment from FAT... *** Warning - bad CRC, using default environment

Loading Environment from MMC... *** Warning - bad CRC, using default environment

<ethaddr> not set. Validating first E-fuse MAC
Net:   eth0: ethernet@4a100000
Warning: usb_ether MAC addresses don't match:
Address in ROM is               de:ad:be:ef:00:01
Address in environment is       04:a3:16:b8:0a:0f
, eth1: usb_ether
Press SPACE to abort autoboot in 2 seconds
=> setenv bootargs console=ttyO0,115200n8 quiet root=/dev/mmcblk0p2 ro rootfstype=ext4 rootwait
=> ext2load mmc 0:2 82000000 /boot/image.fit
11241590 bytes read in 981 ms (10.9 MiB/s)
=> bootm 82000000
## Loading kernel from FIT Image at 82000000 ...
   Using 'conf@1' configuration
   Verifying Hash Integrity ... sha1,rsa2048:dev+ OK
   Trying 'kernel@1' kernel subimage
     Description:  unavailable
     Created:      2020-04-27  22:54:28 UTC
     Type:         Kernel Image
     Compression:  lzo compressed
     Data Start:   0x820000a8
     Data Size:    11198460 Bytes = 10.7 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: 0x80008000
     Entry Point:  0x80008000
     Hash algo:    sha1
     Hash value:   aaddef8a5a4bc96662774d4099f5ebed669d192d
   Verifying Hash Integrity ... sha1+ OK
## Loading fdt from FIT Image at 82000000 ...
   Using 'conf@1' configuration
   Verifying Hash Integrity ... sha1,rsa2048:dev+ OK
   Trying 'fdt@1' fdt subimage
     Description:  beaglebone-black
     Created:      2020-04-27  22:54:28 UTC
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0x82aae18c
     Data Size:    41336 Bytes = 40.4 KiB
     Architecture: ARM
     Hash algo:    sha1
     Hash value:   b4d26307cfaccf57e6e49a8b43028d7c95ebf10d
   Verifying Hash Integrity ... sha1+ OK
   Booting using the fdt blob at 0x82aae18c
   Uncompressing Kernel Image
Image too large: increase CONFIG_SYS_BOOTM_LEN: 0x1000000
Required Size minimum: 0x1000000
Must RESET board to recover
resetting ...
```
The size of the kernel is: 

```
# du -k Image 
18088	Image
```

I have tried to compile with both `make bb.org_defconfig` and `multi_v7_defconfig`.

Also, I have tried to increase the size of `CONFIG_SYS_BOOTM_LEN` from `u-boot/common/bootm.c`  but nothing seems to work. 

Currently it's set to

```
#define CONFIG_SYS_BOOTM_LEN    0x1A7D8C0
```
Can you please help me out here?

Thanks,

RobertCNelson, 2020-04-27:
```
U-Boot SPL 2013.04-dirty (Jul 10 2013 - 14:02:53)
U-Boot 2020.04-00684-gd202f67db0-dirty (Apr 27 2020 - 18:54:58 -0400)
```
Your version of U-Boot SPL and U-Boot don't match.. You should really fix that..

Sorry, i've never tried verified boot, not sure i'll be too helpful...

meghaviparikh, 2020-04-28:
Hi @RobertCNelson,

Do you have any documentation on how do I fix that? I really have been just doing, 

```
git clone https://github.com/u-boot/u-boot.git
make am335x_boneblack_vboot_config O=b/am335x_boneblack_vboot
make all O=b/am335x_boneblack_vboot
``` 
Is there any mistake in here?

Thanks,

@meghaviparikh

RobertCNelson, 2020-04-28:
I use the generifc "am335x_evm_defconfig" and dd the files to the media:

```
make ARCH=arm CROSS_COMPILE=${CC} distclean
make ARCH=arm CROSS_COMPILE=${CC} am335x_evm_defconfig
make ARCH=arm CROSS_COMPILE=${CC}
```

```
sudo dd if=./u-boot/MLO of=${DISK} count=1 seek=1 bs=128k
sudo dd if=./u-boot/u-boot.img of=${DISK} count=2 seek=1 bs=384k
```
REgards,

```

meghaviparikh, 2020-04-28:
Thank you for sharing this, I was able to update the SPL to latest version, although this still didn't solve my problem. Also, I tried to modify the code in `common/bootm.c` of Uboot: 

https://github.com/u-boot/u-boot/blob/d16d37bcd4087b8ea0f66cb76a73edad182d151a/common/bootm.c#L322

```C
printf("Image too large: increase CONFIG_SYS_BOOTM_LEN: 0x%lx\nRequired Size minimum: 0x%lx\n", CONFIG_SYS_BOOTM_LEN, uncomp_size);
```

And tried to print the size that was required. Although I had set a high value for this (as mentioned above):

```C
#define CONFIG_SYS_BOOTM_LEN    0x1A7D8C0
```

I see the following output: 

```
Image too large: increase CONFIG_SYS_BOOTM_LEN: 0x1000000
Required Size minimum: 0x1000000
Must RESET board to recover
```
I wonder why this is so?

Thanks,

@meghaviparikh

RobertCNelson, 2020-04-28:
i'd ask the u-boot developers: https://lists.denx.de/listinfo/u-boot

meghaviparikh, 2020-04-28:
Hi @RobertCNelson,

Thanks for your reply. I'll get in touch with the u-boot developers. Meanwhile, which kernel image do you recommend to use that has size below 15-16 MB? I am only looking for basic Utilities like WiFi and Bluetooth and Ethernet. 

Thanks,

@meghaviparikh

RobertCNelson, 2020-04-28:
Sorry, feel free to customize the example kernel configuration to fit your needs.

closed 2020-04-28 by RobertCNelson