beagleboard/linux #232: Difference between this kernel and TI one

MagnaboscoL, 2020-05-12:
Hi all,

I am wondering which are the main differences between this kernel and the kernel provided by TI.

At the moment I am using Yocto with [meta-ti](https://git.yoctoproject.org/cgit/cgit.cgi/meta-ti/) layer that pulls the kernel from git://git.ti.com/ti-linux-kernel/ti-linux-kernel.git branch *ti-rt-linux-5.4.y*.

I see that boot2qt was using this repository for the kernel, instead of TI one before dropping beaglebone support (see [https://code.qt.io](https://code.qt.io/cgit/yocto/meta-boot2qt.git/diff/meta-ti-extras/recipes/linux/linux-ti-staging_%25.bbappend?id=17747110065d6d1482fb5786773c27d78f7c8bac)) so I guess that mixing this kernel with meta-ti drivers would be possible if needed.

RobertCNelson, 2020-05-12:
@MagnaboscoL 

https://github.com/RobertCNelson/ti-linux-kernel-dev/tree/ti-linux-5.4.y

It's ti 5.4.x-ti git c8469270aa0e018913e0af1de913a70257742e6e plus all our BeagleBoard.org changes (u-boot overlays, brcmfmac patches, Wireguard, etc..)  We try to sync up with ti about once a week...

Regards,

MagnaboscoL, 2020-05-12:
@RobertCNelson thanks a lot for your quick reply!

I will try to use your branch then.

I see that Linaro toolchain is somehow recommended. Is it compulsory to use that one instead of the default yocto one (at the moment named arm-poky-linux-gnueabi)

Regards
Luca

RobertCNelson, 2020-05-12:
"pre-built" compilers are recommend, as they are really easy to install on a "new" user's system..

Use whatever compiler you want..

Side note, i'd love to get a pre-built llvm toolchain working. ;)

Regards,

MagnaboscoL, 2020-05-14:
Ok, thanks @RobertCNelson,

anyway I am still missing something that is probably obvious.
I try to rephrase just to double check:

https://github.com/RobertCNelson/ti-linux-kernel-dev/tree/ti-linux-5.4.y is a set of scripts that pulls the kernel from the TI Kernel mirror (https://github.com/RobertCNelson/ti-linux-kernel.git) and applies a set of patches for: drivers, RT, WireGuard, ...

The resulting kernel is a patched version of the TI one and do NOT include the kernel modules for the GPU, PRU, ... that can be built separately (as in [meta-ti](https://git.yoctoproject.org/cgit/cgit.cgi/meta-ti/) Yocto layer). Right?

On the other hand https://github.com/beagleboard/linux is somehow "more complete" and already includes GPU, PRU kernel modules. Right?

If what is written above is correct:

- would be correct to suppose that using https://github.com/beagleboard/linux you can use TI SOC specific features like SGX530 GPU and the PRUs?
- "OpenGL ES2" would be available compiling [ti-sgx-ddk-um](https://git.ti.com/graphics/omap5-sgx-ddk-um-linux). Right? (Somehow similar too the old [SGX/OpenGL Status
](https://github.com/beagleboard/linux/issues/31))

Regards

RobertCNelson, 2020-05-14:
Incorrect, https://github.com/beagleboard/linux and https://github.com/RobertCNelson/ti-linux-kernel-dev/tree/ti-linux-5.4.y are exactly the same..

On every "release" of https://github.com/RobertCNelson/ti-linux-kernel-dev/ the "KERNEL" directory get's pushed to https://github.com/beagleboard/linux and tagged..

As far as SGX530 and PRU:

SGX530: When TI enables this, we get it automatically.

PRU: TI uses remoteproc, we give our users an option: UIO or remoteproc, so we remove just enough from the *.dts to make both options work (as a separate u-boot overlay)..

I haven't done 5.4.x-ti pru yet, so this is 4.19.x-ti for comparison:

v4.19.x-ti remoteproc: https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/AM335X-PRU-RPROC-4-19-TI-00A0.dts
Generic UIO Overlay: https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/AM335X-PRU-UIO-00A0.dts

SGX Drivers: yeah you'll need to build the matching ti-sgx-ddk-um..

Regards,

MagnaboscoL, 2020-05-17:
Ok, thanks a lot for the clear and detailed reply.

Regards,

closed 2020-05-17 by MagnaboscoL