beagleboard/linux #233: kernel BUG at kernel/sched/deadline ?

MagnaboscoL, 2020-05-21:
Hi,

I am just starting to test a "custom distro" I have made using this kernel and Yocto.
I am facing a "kernel BUG" I cannot understand.

Using:

- kernel 5.4-rt [1fe65cd](https://github.com/beagleboard/linux/commit/1fe65cd99d9e6b466c37ffda016fd65d1f91a0bd)
- defconfig from [ti-linux-rt-5.4.y/patches/defconfig](https://raw.githubusercontent.com/RobertCNelson/ti-linux-kernel-dev/ti-linux-rt-5.4.y/patches/defconfig). The only modifications I have done are unsetting both `CONFIG_MODULE_COMPRESS` and `CONFIG_MODULE_COMPRESS_XZ`
- kernel module omap5-sgx-ddk for GPU support from [https://git.ti.com](https://git.ti.com/cgit/graphics/omap5-sgx-ddk-linux/log/?h=ti-img-sgx/1.17.4948957/k5.4)
- other packages from [meta-ti](https://git.yoctoproject.org/cgit/cgit.cgi/meta-ti/tag/?h=ti2020.00) yocto layer (tag: [ti2020.00](https://git.yoctoproject.org/cgit/cgit.cgi/meta-ti/tag/?h=ti2020.00))
- [BB-BONE-LCD5-01-00A1.dts](https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/BB-BONE-LCD5-01-00A1.dts)

Hardware:

- BeagleBone green
- a resistive LCD [Seeed-Studio-BeagleBone-Green-LCD-Cape](https://wiki.seeedstudio.com/Seeed-Studio-BeagleBone-Green-LCD-Cape-with-Resistive-Touch/)

When I start Weston touch screen calibration I get:

```
[   63.286156] 000: kernel BUG at kernel/sched/deadline.c:1495!
[   63.291842] 000: Internal error: Oops - BUG: 0 [#1] PREEMPT_RT SMP ARM
[   63.298400] 000: Modules linked in:
[   63.301902] 000:  pvrsrvkm(O)
[   63.304877] 000:  joydev
[   63.307415] 000:  evdev
[   63.309866] 000:
[   63.311798] 000: CPU: 0 PID: 644 Comm: weston Tainted: G           O      5.4.20 #1
[   63.319491] 000: Hardware name: Generic AM33XX (Flattened Device Tree)
[   63.326044] 000: PC is at enqueue_task_dl+0x90/0xd8c
[   63.331054] 000: LR is at rt_mutex_setprio+0x358/0x568
[   63.336223] 000: pc : [<c018c014>]    lr : [<c01754d4>]    psr: 20000093
[   63.342950] 000: sp : da4afa58  ip : da4afb20  fp : da4afb1c
[   63.348630] 000: r10: df926140  r9 : dc32c420  r8 : 00000001
...
```

Pleased find attached the entire log
[bug_log.txt](https://github.com/beagleboard/linux/files/4664292/bug_log.txt)

Note: attached log has been captured when starting the touch screen calibration, but I got the very same bug with an image without Weston at the end of `ts_calibrate` from [tslib](https://github.com/libts/tslib).

I suppose that the issue is independent form the applications I was runnig and maybe related to some wrong compilation settings (that probably influenced the module pvrsrvkm.ko), but for the moment I am quite clueless.

Any hint would be very appreciated.

Regards
Luca

Pillar1989, 2020-06-11:
@pdp7  not yet. we tested it on version 4.19.

MagnaboscoL, 2020-06-12:
Hi @pdp7,

I have a Yocto setup that seems to work* with this LCD Cape.
At the moment I am using TI kernel and I could switch kernel for testing.
Please let me know if you think I could do something to speed up the testing.

* Just to give complete info: It seems to work but I am still facing a color issue when using the GPU for QT apps as described in [e2e.ti.com](https://e2e.ti.com/support/processors/f/791/p/913245/3376011#3376011). Anyway that could probably be unrelated to the tests needed for this issue.

Regards