beagleboard/linux #239: remoteproc write to PRU over rpmsg device blocks even when set non-blocking

buzmeg, 2020-06-30:
When the PRU buffers fill up, they block for 15 seconds and then fail with ERESTARTSYS.

This is ... sorta okay ... when in blocking mode.  However, if O_NONBLOCK is set, the system should immediately come back with ENOMEM on an attempt to write when the PRU buffers are full.

The problem seems to be that rpmsg_pru_write needs to call rpmsg_trysend when O_NONBLOCK is set.

See: rpmsg_eptdev_write_iter line 244 in rpmsg_char.c for an example:
https://github.com/beagleboard/linux/blob/38b462b53e1ce18b0727a6852c8c87dd96f56f6f/drivers/rpmsg/rpmsg_char.c#L244

See this Google Groups thread for more discussion:
https://groups.google.com/forum/#!category-topic/beagleboard/6Ch7Do4Hm7k

Thanks.

pdp7, 2020-07-08:
@MarkAYoder have you run into this PRU issue?

buzmeg, 2020-07-10:
Yes.  I'm the one who hit this in the original thread.

MarkAYoder, 2020-07-10:
@pdp7 I haven't seen this issue.  Sorry.

buzmeg, 2020-10-02:
Any resolution on this?

buzmeg, 2020-11-20:
Bumping this two months later.

buzmeg, 2021-05-22:
This is still sitting.

closed 2021-05-27 by RobertCNelson