beagleboard/linux #245: How disable i2c-1 once we are using CAN on PocketBeagle

juvinski, 2020-08-16:
Hi guys,

there is a way to disable/remove i2c1 from pocket beagle ?
We are using the CAN interface on pocketbeagle and they share de same pins, and the I2C1 is appearing, what is causing problems once this i2c is not in use but exists in the filesystem.
On black and blue, once there is a DTC the i2c1 is not present, but on pocket with kernel 4.19 is present.

RobertCNelson, 2020-08-16:
Just remove:

https://github.com/beagleboard/BeagleBoard-DeviceTrees/blob/v4.19.x-ti/src/arm/am335x-pocketbeagle.dts#L1050-L1056

CHange the default line to can:
https://github.com/beagleboard/BeagleBoard-DeviceTrees/blob/v4.19.x-ti/src/arm/am335x-pocketbeagle.dts#L616-L636

and

https://github.com/beagleboard/BeagleBoard-DeviceTrees/blob/v4.19.x-ti/src/arm/am335x-pocketbeagle.dts#L652-L672

make
sudo make install
sudo reboot

that repo should be under /opt/source/dtb-4.19/

Regards,

juvinski, 2020-08-16:
Hi @RobertCNelson 

thanks for the tips, I will try and update you.

Thanks a lot.

RobertCNelson, 2020-08-16:
Also there are two overlays for this:

https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/PB-CAN0-00A0.dts

https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/PB-CAN1-00A0.dts

https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Override_external_capes

Regards,

juvinski, 2020-08-16:
Hi @RobertCNelson ,

I'm working on a new version of PockePilot cape for PocketBeagle + ardupilot, using config-pin the can devices are working without loading any overlay at uBoot but I will give a shot on this true.

thanks again for your help.