beagleboard/linux #248: Outdated dtc version

amilcarlucas, 2020-08-26:
Hello,

on the `https://rcn-ee.com/rootfs/2020-04-09/elinux/ubuntu-18.04.4-console-armhf-2020-04-09.tar.xz` image the installed dtc version is:

```
dtc --version
Version: DTC 1.4.5
```

And system information from sudo /opt/scripts/tools/version.sh :

```
git:/opt/scripts/:[7cdc270818b76d829d247cf05fe309c320a3929a]
eeprom:[A335BNLT00C03318BBBK0715]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[rcn-ee.net Ubuntu Console Image 2020-04-09]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2019.04-00002-gc9b3922522]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-UIO-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
UBOOT: Loaded Overlay:[BB-HDMI-TDA998x-00A0]
UBOOT: Loaded Overlay:[BBORG_COMMS-00A2]
UBOOT: Loaded Overlay:[BBORG_PROTO-00A2]
UBOOT: Loaded Overlay:[BBORG_RELAY-00A2]
kernel:[4.19.94-ti-r42]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200403.0-0rcnee0~bionic+20200403]
pkg:[bb-wl18xx-firmware]:[1.20200322.0-0rcnee0~bionic+20200322]
pkg:[kmod]:[24-1ubuntu3.2rcnee0~bionic+20190208]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
groups:[ubuntu : ubuntu adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait uboot_detected_capes=BBORG_RELAY,BBORG_COMMS,BBORG_PROTO, coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
[   55.622110] remoteproc remoteproc0: wkup_m3 is available
[   55.628157] remoteproc remoteproc0: powering up wkup_m3
[   55.628186] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[   55.628438] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
dmesg | grep pinctrl-single
[    0.818928] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    0.820287] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 002: ID 0781:5591 SanDisk Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```

But the version is not supported by the overlays, It requires at least 1.4.7 and it explicitly says to not use 1.4.5 .

Can you please fix this?

RobertCNelson, 2020-08-26:
image: 2020-04-09

bb.org-overlays: 2020-04-30: dtc 1.4.7 is required
https://github.com/beagleboard/bb.org-overlays/commit/0e3c94072dd98c1cdd877e773be0e68f67479acb

device-tree-compiler: 1.4.7-3ubuntu2rcnee0: pushed on 2020-04-30

> Can you please fix this?

Already did, feel free to run:

```
sudo apt update
sudo apt upgrade
```

Regards,

amilcarlucas, 2020-08-26:
Nice, thanks!!!!
that did the trick.

closed 2020-08-26 by amilcarlucas