beagleboard/linux #249: Using kernel 4.14 and buildroot for building the rootfs, how can I use the sgx530 graphics accelerator?

phantom1299, 2020-09-18:
**tl;dr:**
Is it necessary for me to change the kernel version or repository from branch `4.14` to be able to use the sgx530?

I'm currently working on a project which uses a custom board which is mostly identical to the Beaglebone Black (same SoC).

I'm building the bootloader (u-boot), the kernel (branch 4.14 in this repo) and the rootfs as initramfs (using buildroot with busybox) separately. I'm using beaglebones dts files inside the kernels source files and compile them into a dtb file. 

When my board boots up, it loads the rootfs into ram and does everything there. So I can't add anything to my rootfs at runtime. I have to install the neccesary modules at buildtime.

When I run  `ls /sys/devices/platform/ocp/56000000.sgx` the output is as follows:
```
ls /sys/devices/platform/ocp/56000000.sgx
driver_override  of_node          subsystem
modalias         power            uevent
```

I first tried using the TI Graphics SDK folowing the [User Guide](https://processors.wiki.ti.com/index.php/Graphics_SDK_Quick_installation_and_user_guide) but got compilation errors like:

```
 make BUILD=release OMAPES=8.x all
...
/Graphics_SDK_5_01_01_02/GFX_Linux_KM/services4/srvkm/env/linux/mmap.c:778:47: error: incompatible type for argument 3 of ‘vm_insert_mixed’
    result = vm_insert_mixed(ps_vma, ulVMAPos, pfn);
                                               ^~~
In file included from /home/muhammed/Graphics_SDK_5_01_01_02/GFX_Linux_KM/services4/srvkm/env/linux/mmap.c:50:0:
./include/linux/mm.h:2380:5: note: expected ‘pfn_t {aka struct <anonymous>}’ but argument is of type ‘IMG_UINTPTR_T {aka long unsigned int}’
```

Is it necessary for me to change the kernel version or repository to be able to use the sgx530?

RobertCNelson, 2020-09-18:
@phantom1299 , for that kernel you need this branch for the kernel modules:

https://git.ti.com/gitweb?p=graphics/omap5-sgx-ddk-linux.git;a=shortlog;h=refs/heads/ti-img-sgx/1.14.3699939/k4.14

Here is how i build it, two patches for debian: https://github.com/rcn-ee/ti-sgx-omap5/blob/master/build-ti335x-4.14.x.sh

and use this repo for the blobs:

https://git.ti.com/gitweb?p=graphics/omap5-sgx-ddk-um-linux.git;a=summary

I use this git checkout:

```
git_repo="git://git.ti.com/graphics/omap5-sgx-ddk-um-linux.git"
git_sha="bbbd5cbb55e4c54d3b02456ae553bea86fd61506"
```

and then ^ will look for libgbm from here, it clashes with mesa's libgbm..

```
git_repo="git://git.ti.com/glsdk/libgbm.git"
git_sha="f316cf4b98ca3f4a1ca92a48ddd8aa559894a45c"
```

Regards,

phantom1299, 2020-09-18:
@RobertCNelson thank you for the quick reply.  I think I followed the steps correctly but I'll list them to confirm:

Building the km:

1. `git clone https://github.com/rcn-ee/ti-sgx-omap5.git`
2. `cd ti-sgx-omap5/src`
3. `export KERNELDIR=/path/to/linux`
4. `../build-ti335x-4.14.x.sh`
5. output:
```
  ...
  CC      ti-sgx-omap5/src/src/eurasia_km/eurasiacon/binary2_omap_linux_release/target/kbuild/pvrsrvkm.mod.o
  LD [M]  ti-sgx-omap5/src/src/eurasia_km/eurasiacon/binary2_omap_linux_release/target/kbuild/pvrsrvkm.ko
```
The output file is `pvrsrvkm.ko` which i will later install with insmod on my custom board.

Building the um:

1. `git clone https://git.ti.com/git/graphics/omap5-sgx-ddk-um-linux.git`
2. `cd omap5-sgx-ddk-um-linux/`
3. `git checkout bbbd5cbb55e4c54d3b02456ae553bea86fd61506`
4. `export DISCIMAGE=/path/to/my/target`
5. `export TARGET_PRODUCT=ti335x`
6. `sudo -E make install`
7. output:
```
...
cp -ar ./targetfs/lib/*/path/to/my/target/usr/lib
cp -ar ./targetfs/include/*/path/to/my/target/usr/include
cp -ar ./targetfs/bin/* /path/to/my/target/usr/bin
cp -ar ./targetfs/share/* /path/to/my/target/usr/share
```
I rebuild my target and setup my custom board. In it I run:
```
 # insmod pvrsrvkm.ko
 pvrsrvkm: loading out-of-tree module taints kernel.
```

Now, If I didn't do any mistakes I just need to install libgbm:
```
# pvrsrvctl
pvrsrvctl: error while loading shared libraries: libgbm.so.2: cannot open shared object file: No such file or directory
```

RobertCNelson, 2020-09-18:
Yeap you got it. ;)

PS, here is a reference image you can use tor comparison, SGX enabled out of the box..

https://debian.beagleboard.org/images/bone-debian-9.12-imgtec-armhf-2020-04-06-4gb.img.xz

Regards,

phantom1299, 2020-09-18:
I was going to give up on this. Thank god I asked and thank you for helping. I realy appreciate it.

closed 2020-09-18 by phantom1299