beagleboard/linux #250: 5.4: Unable to create debian packages with make deb-pkg

JOO200, 2020-09-18:
I'm unable to create a debian package for the kernel 5.4 branch.

I cloned the repository and I locally have the 5.4 branch.

I use the following commands on the command line. 
`make ARCH=arm defconfig`
`make ARCH=arm -k -j4 CROSS_COMPILE=arm-linux-gnueabihf- deb-pkg`
I double checked that all files are clean (git reset --hard 5.4 and git clean -xf)

This is the build log: https://pastebin.com/xhKpMiyb Pls ignore the perl warnings, I changed the console language to english so you don't have to have some weird german messages.

Thanks for your help.

RobertCNelson, 2020-09-18:
Hi @JOO200 

Switch to "bindeb-pkg"

```
make ARCH=arm -k -j4 CROSS_COMPILE=arm-linux-gnueabihf- bindeb-pkg
```

deb-pkg tries to generate teh source, but then bomb's on the am335x firmware blobs..

Regards,

JOO200, 2020-09-18:
Oh, thanks for the quick response. That solved the issue.

closed 2020-09-18 by JOO200