beagleboard/linux #251: No such pinmux file

Bence98007, 2020-10-17:
Hi there!

I have a problem configuring a pin P2_18 and P2_06 on PocketBeagle.
What I'm trying is to get SWCAN working, but I have to set it up to HIGH.
When I run config-pin `P2_18 `hi it returns error:
```
config-pin P2_18 hi
ERROR: open() for /sys/devices/platform/ocp/ocp:P2_18_pinmux/state failed, No such file or directory
```

I found no P2_18 or P2_06 pinmux file exists here:
[image]

version.sh output:

`
root@beaglebone:~# /opt/scripts/tools/version.sh
git:/opt/scripts/:[7cdc270818b76d829d247cf05fe309c320a3929a]
eeprom:[A335PBGL00A21743GPB30299]
model:[TI_AM335x_PocketBeagle]
dogtag:[BeagleBoard.org Debian Buster IoT Image 2020-08-25]
bootloader:[microSD]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-gc9b3922522]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-pocketbeagle.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-19-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[PB-CAN0-00A0]
UBOOT: Loaded Overlay:[PB-CAN1-00A0]
UBOOT: Loaded Overlay:[PB-MCP2515-SPI1]
kernel:[4.19.94-ti-r50]
nodejs:[v10.21.0]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_addr4=/lib/firmware/PB-CAN0-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr5=/lib/firmware/PB-CAN1-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr6=/lib/firmware/PB-MCP2515-SPI1.dtbo]
uboot_overlay_options:[uboot_overlay_pru=AM335X-PRU-RPROC-4-19-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200814.0-0~buster+20200814]
pkg:[bb-wl18xx-firmware]:[1.20200813.1-0~buster+20200813]
pkg:[kmod]:[26-1]
pkg:[librobotcontrol]:[1.0.5-git20200715.0-0~buster+20200716]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal input bluetooth netdev i2c gpio admin spi iio docker tisdk weston-launch xenomai cloud9ide pwm eqep remoteproc]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
[   53.832040] remoteproc remoteproc0: wkup_m3 is available
[   54.008667] remoteproc remoteproc0: powering up wkup_m3
[   54.008700] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[   54.008997] remoteproc remoteproc0: remote processor wkup_m3 is now up
[   55.838857] remoteproc remoteproc1: 4a334000.pru is available
[   55.872421] remoteproc remoteproc2: 4a338000.pru is available
dmesg | grep pru
[   55.838857] remoteproc remoteproc1: 4a334000.pru is available
[   55.839046] pru-rproc 4a334000.pru: PRU rproc node pru@4a334000 probed successfully
[   55.872421] remoteproc remoteproc2: 4a338000.pru is available
[   55.872630] pru-rproc 4a338000.pru: PRU rproc node pru@4a338000 probed successfully
dmesg | grep pinctrl-single
[    0.895597] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    0.905153] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 002 Device 002: ID 148f:7601 Ralink Technology, Corp. MT7601U Wireless Adapter
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
`

What can cause this error?
I tried to run pinmux-generator, but it throws error:

```
root@beaglebone:/opt/source/bb.org-overlays/tools/pinmux-generator# ./PocketBeagle.sh
pinmux-4.0.1526-setup.run                                            100%[=====================================================================================================================================================================>]  39.87M  2.91MB/s    in 14s

2020-10-17 15:20:15 (2.76 MB/s) - ‘pinmux-4.0.1526-setup.run’ saved [41810329/41810329]

./lib/pinmux.sh: line 220: ./pinmux-4.0.1526-setup.run: cannot execute binary file: Exec format error
cp: cannot stat 'tmp/pinmux/deviceData/AM335x/AM335x.json': No such file or directory
##################
P1_02
cat: AM335x.json: No such file or directory
cat: AM335x.json: No such file or directory
cat: AM335x.json: No such file or directory
cat: AM335x.json: No such file or directory

```

Tried to install pinmux manually, but it doesnt support armhf architecture.
Can anyone help me with this?

Bence98007, 2020-10-18:
Anyone?

RobertCNelson, 2020-10-19:
Sure..

Those pins are utilized by the overlay you loaded

https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/PB-MCP2515-SPI1.dts

Regards,

Bence98007, 2020-10-19:
Hi @RobertCNelson 

So I cannot use the config-pin command on that? Or how should I set it to HIGH?

RobertCNelson, 2020-10-19:
config-pin is only for pins not configured..  If you want to set the pin high, change the overlay..

Regards,

Bence98007, 2020-10-19:
> config-pin is only for pins not configured.. If you want to set the pin high, change the overlay..
> 
> Regards,

Thanks for the quick answer! 👍  
What do You mean by change the overlay?

Bence98007, 2020-10-19:
nvm, made it work.

closed 2020-10-19 by Bence98007