beagleboard/linux #255: Error. Unable to build 5.4-rt kernel for BBB on BBB

bernid, 2020-11-07:
Hi, 
I'm trying to build 5.4.70-ti-rt-r19 kernel on my BBB. 
During check out I have received warnings: 'the following paths have collided':
```
debian@beaglebone:/opt/di$ sudo git clone -b 5.4-rt --depth 1 https://github.com/beagleboard/linux.git
Cloning into 'linux'...
remote: Enumerating objects: 70276, done.
remote: Counting objects: 100% (70276/70276), done.
remote: Compressing objects: 100% (66104/66104), done.
remote: Total 70276 (delta 6002), reused 14268 (delta 3326), pack-reused 0
Receiving objects: 100% (70276/70276), 186.48 MiB | 919.00 KiB/s, done.
Resolving deltas: 100% (6002/6002), done.
Checking out files: 100% (66180/66180), done.
warning: the following paths have collided (e.g. case-sensitive paths
on a case-insensitive filesystem) and only one from the same
colliding group is in the working tree:

  'include/uapi/linux/netfilter/xt_CONNMARK.h'
  'include/uapi/linux/netfilter/xt_connmark.h'
  'include/uapi/linux/netfilter/xt_DSCP.h'
  'include/uapi/linux/netfilter/xt_dscp.h'
  'include/uapi/linux/netfilter/xt_MARK.h'
  'include/uapi/linux/netfilter/xt_mark.h'
  'include/uapi/linux/netfilter/xt_RATEEST.h'
  'include/uapi/linux/netfilter/xt_rateest.h'
  'include/uapi/linux/netfilter/xt_TCPMSS.h'
  'include/uapi/linux/netfilter/xt_tcpmss.h'
  'include/uapi/linux/netfilter_ipv4/ipt_ECN.h'
  'include/uapi/linux/netfilter_ipv4/ipt_ecn.h'
  'include/uapi/linux/netfilter_ipv4/ipt_TTL.h'
  'include/uapi/linux/netfilter_ipv4/ipt_ttl.h'
  'include/uapi/linux/netfilter_ipv6/ip6t_HL.h'
  'include/uapi/linux/netfilter_ipv6/ip6t_hl.h'
  'net/netfilter/xt_DSCP.c'
  'net/netfilter/xt_dscp.c'
  'net/netfilter/xt_HL.c'
  'net/netfilter/xt_hl.c'
  'net/netfilter/xt_RATEEST.c'
  'net/netfilter/xt_rateest.c'
  'net/netfilter/xt_TCPMSS.c'
  'net/netfilter/xt_tcpmss.c'
  'tools/memory-model/litmus-tests/Z6.0+pooncelock+poonceLock+pombonce.litmus'
  'tools/memory-model/litmus-tests/Z6.0+pooncelock+pooncelock+pombonce.litmus'
```
And at the end after few hours of building received an error related to one of the above directory:
```
sudo make bindeb-pkg
....
....
ble-pointer-types]
   if (!set_ect_tcp(skb, einfo))
                         ^~~~~
net/ipv4/netfilter/ipt_ECN.c:45:61: note: expected 'const struct ipt_ECN_info *' but argument is of type 'const struct ipt_ECN_info *'
 set_ect_tcp(struct sk_buff *skb, const struct ipt_ECN_info *einfo)
                                  ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
net/ipv4/netfilter/ipt_ECN.c: In function 'ecn_tg_check':
net/ipv4/netfilter/ipt_ECN.c:98:11: error: dereferencing pointer to incomplete type 'const struct ipt_ECN_info'
  if (einfo->operation & IPT_ECN_OP_MASK)
           ^~
net/ipv4/netfilter/ipt_ECN.c:98:25: error: 'IPT_ECN_OP_MASK' undeclared (first use in this function); did you mean 'IPT_ECN_IP_MASK'?
  if (einfo->operation & IPT_ECN_OP_MASK)
                         ^~~~~~~~~~~~~~~
                         IPT_ECN_IP_MASK
net/ipv4/netfilter/ipt_ECN.c:104:27: error: 'IPT_ECN_OP_SET_ECE' undeclared (first use in this function); did you mean 'IPT_ECN_OP_MATCH_ECE'?
  if ((einfo->operation & (IPT_ECN_OP_SET_ECE|IPT_ECN_OP_SET_CWR)) &&
                           ^~~~~~~~~~~~~~~~~~
                           IPT_ECN_OP_MATCH_ECE
net/ipv4/netfilter/ipt_ECN.c:104:46: error: 'IPT_ECN_OP_SET_CWR' undeclared (first use in this function); did you mean 'IPT_ECN_OP_MATCH_CWR'?
  if ((einfo->operation & (IPT_ECN_OP_SET_ECE|IPT_ECN_OP_SET_CWR)) &&
                                              ^~~~~~~~~~~~~~~~~~
                                              IPT_ECN_OP_MATCH_CWR
net/ipv4/netfilter/ipt_ECN.c: At top level:
net/ipv4/netfilter/ipt_ECN.c:116:23: error: invalid application of 'sizeof' to incomplete type 'struct ipt_ECN_info'
  .targetsize = sizeof(struct ipt_ECN_info),
                       ^~~~~~
cc1: some warnings being treated as errors
make[6]: *** [scripts/Makefile.build:266: net/ipv4/netfilter/ipt_ECN.o] Error 1
make[5]: *** [scripts/Makefile.build:500: net/ipv4/netfilter] Error 2
make[4]: *** [scripts/Makefile.build:500: net/ipv4] Error 2
make[3]: *** [Makefile:1732: net] Error 2
make[2]: *** [debian/rules:6: build] Error 2
dpkg-buildpackage: error: debian/rules build subprocess returned exit status 2
make[1]: *** [scripts/Makefile.package:83: bindeb-pkg] Error 2
make: *** [Makefile:1467: bindeb-pkg] Error 2
```
How to solve this issue? Thanks

RobertCNelson, 2020-11-07:
Use a native linux file system.

Fat/ntfs or what ever your using for that drive doesn't support case sensitive.

Regards

bernid, 2020-11-09:
> Use a native linux file system.
> 
> Fat/ntfs or what ever your using for that drive doesn't support case sensitive.
> 
> Regards

Yes I had SD card formatted as FAT mounted to debian on eMMC. 
After format it as ext4 (after about 18h of building) new kernel has been build. 
Thank you.

closed 2020-11-09 by bernid