beagleboard/linux #258: Custom audio cape

Maobuff, 2021-07-15:
Hi, I'm making a custom audio cape for BBB. The goal is to record sound using microphones.
I used tlv320aic3106 codec, modified BB-BONE-AUDI....dts.
Replaced all 3104 with 3106, swapped axr0 with axr2 (because of a mistake in my board). Enabled micbias.
Compiled, enabled it in uEnv.txt.
But there is no audio in the recordings. I can see data movement on mcasp bus on an oscilloscope. And looks like arecord is working fine. If I remove custom cape arecord is stuck (as expected).
arecord show max peak of 0% all the time. In alsamixer PGA Capture Switch turned on, PGA Capture Volume set to 105.

Attaching modified dto. [BB-BONE-AUDI-02-00A0.txt](https://github.com/beagleboard/linux/files/6821214/BB-BONE-AUDI-02-00A0.txt)

miral2525, 2025-01-24:
Here the key change should be done for enhancement is to  swapped mcasp0_axr0 with mcasp0_axr2. Verify that the PIN_INPUT_PULLDOWN | MUX_MODE2 is correct for your hardware. For mcasp0_axr2, the correct mode should be MUX_MODE2, and the other McASP pins should also match their intended functionality.

miral2525, 2025-01-24:
Also some othet stuff should be done so 

[BB-BONE-AUDI-02-00A0 edited by Miral.txt](https://github.com/user-attachments/files/18534578/BB-BONE-AUDI-02-00A0.edited.by.Miral.txt)