beagleboard/linux #259: ttyACM device is created very late

epsilontheta, 2021-08-05:
**Describe the bug**
After booting the Beaglebone I see in dmesg that the USB device is detected properly after 4 seconds.
Then it takes another 36 seconds until the ttyACM0 device is created under /dev.
This is a problem if you have a service that expects the device to be created when it starts.
It seems like this is a regression w.r.t. the Debian 9 image as this delay was not present back then.

```
[    3.659645] usb 1-1: new full-speed USB device number 2 using musb-hdrc
[    3.808575] usb 1-1: New USB device found, idVendor=1546, idProduct=01a9, bcdDevice= 1.00
[    3.808591] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    3.808599] usb 1-1: Product: u-blox GNSS receiver
[    3.808606] usb 1-1: Manufacturer: u-blox AG - www.u-blox.com
```

```
[   40.782413] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[   40.826035] usbcore: registered new interface driver cdc_acm
[   40.826051] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
```

When replugging the device the ttyACM0 device is created instantly.

**Describe how to reproduce the bug**
Plugin USB device before boot and turn on power.

**REQUIRED INFORMATION**
Run this command and paste the output here:
```git:/opt/scripts/:[b39ec679648a6be8f25f48bd1c9784c1fc5a0c46]
eeprom:[A335BNLTBBG1BBG219102075]
model:[TI_AM335x_BeagleBone_Green]
dogtag:[BeagleBoard.org Debian Buster IoT Image 2020-04-06]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2019.04-00002-g07d5700e21]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot-univ.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-19-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
UBOOT: Loaded Overlay:[BB-I2C1-00A0]
UBOOT: Loaded Overlay:[BB-I2C2-00A0]
UBOOT: Loaded Overlay:[BB-SPIDEV1-00A0]
UBOOT: Loaded Overlay:[BB-UART4-00A0]
UBOOT: Loaded Overlay:[BB-UART5-00A0]
UBOOT: Loaded Overlay:[M-BB-BBG-00A0]
kernel:[4.19.94-ti-r42]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_addr0=/lib/firmware/BB-UART4-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr1=/lib/firmware/BB-UART5-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr2=/lib/firmware/BB-I2C1-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr3=/lib/firmware/BB-I2C2-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr4=/lib/firmware/BB-SPIDEV1-00A0.dtbo]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=0]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20210416.0-0~buster+20210416]
pkg:[bb-wl18xx-firmware]:[1.20210520.0-0~buster+20210520]
pkg:[kmod]:[26-1]
pkg:[librobotcontrol]:[1.0.5-git20200715.0-0~buster+20200716]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
[   50.871253] remoteproc remoteproc0: wkup_m3 is available
[   51.207736] remoteproc remoteproc0: powering up wkup_m3
[   51.207768] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[   51.208251] remoteproc remoteproc0: remote processor wkup_m3 is now up
[   53.639686] remoteproc remoteproc1: 4a334000.pru is available
[   53.642001] remoteproc remoteproc2: 4a338000.pru is available
dmesg | grep pru
[   53.639686] remoteproc remoteproc1: 4a334000.pru is available
[   53.639852] pru-rproc 4a334000.pru: PRU rproc node pru@4a334000 probed successfully
[   53.642001] remoteproc remoteproc2: 4a338000.pru is available
[   53.642172] pru-rproc 4a338000.pru: PRU rproc node pru@4a338000 probed successfully
dmesg | grep pinctrl-single
[    0.941500] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    0.954776] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 002: ID 1546:01a9 U-Blox AG 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```

```
root@beaglebone:~# systemd-analyze blame
         31.960s generic-board-startup.service
         31.175s dev-mmcblk1p1.device
          4.066s loadcpufreq.service
          3.256s systemd-udev-trigger.service
          3.150s custom_a.service
          2.851s networking.service
          2.282s connman.service
          2.121s avahi-daemon.service
          2.013s ssh.service
          1.993s systemd-logind.service
          1.820s wpa_supplicant.service
          1.413s systemd-journald.service
          1.325s systemd-timesyncd.service
          1.277s dnsmasq.service
          1.136s custom_b.service
           916ms systemd-journal-flush.service
           915ms cpufrequtils.service
           898ms systemd-user-sessions.service
           896ms custom_c.service
           593ms user@0.service
           578ms systemd-update-utmp.service
           571ms systemd-remount-fs.service
           535ms save-time.service
           479ms sys-kernel-debug.mount
           446ms dev-mqueue.mount
           440ms systemd-tmpfiles-setup.service
           436ms systemd-modules-load.service
           429ms systemd-udevd.service
           411ms systemd-tmpfiles-setup-dev.service
           402ms systemd-update-utmp-runlevel.service
           396ms fake-hwclock.service
           382ms sys-fs-fuse-connections.mount
           382ms systemd-sysusers.service
           363ms systemd-sysctl.service
           334ms systemd-random-seed.service
           310ms kmod-static-nodes.service
           257ms sys-kernel-config.mount
           213ms systemd-tmpfiles-clean.service
           129ms user-runtime-dir@0.service
           115ms ifupdown-pre.service
```

RobertCNelson, 2021-08-05:
Hi @epsilontheta , it's probally built as a module, change it to built-in should come up as soon as usb..

```
lsmod
```

Regards,

epsilontheta, 2021-08-05:
Ah, good catch. I got the delay down to 8 seconds by creating `/etc/modules-load.d/cdc_acm.conf` and adding the module there. Thanks for your fast response!

closed 2021-08-05 by epsilontheta