beagleboard/linux #263: builig external kernel module

bernid, 2021-09-20:
**Describe the bug**
Hi, I receive an error and cannot build external 5.4-rt  kernel module :
`make[3]: *** No rule to make target 'arch/arm/kernel/module.lds'`
I Checked that there is no module.lds file in the current headers directory::
/usr/src/linux-headers-5.4.106/arch/arm
Few months ago I build the same module without an issue and I had:
/usr/src/linux-headers-5.4.70/arch/arm/kernel/module.lds

**Describe how to reproduce the bug**
#I would like to instal botc module:
#https://github.com/coroner21/botic-modules/tree/5.4
#after checkout:
sudo git clone -b 5.4 https://github.com/coroner21/botic-modules.git 
#create missing directories:
sudo mkdir /lib/modules/$(uname -r)/extramodules
sudo mkdir /lib/modules/$(uname -r)/updates
#change directories
cd botic-modules
sudo make

**REQUIRED INFORMATION**
Run this command and paste the output here:
```
git:/opt/scripts/:[f495239e456d8ec99554a35ae04b33ddce5c26cc]
eeprom:[A335BNLT00C05214BBBK099E]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Buster Console Image 2020-07-13]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2019.04-00002-g31a8ae0206]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot.dts]
UBOOT: Loaded Overlay:[BB-ADC-00A0.bb.org-overlays]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0.bb.org-overlays]
UBOOT: Loaded Overlay:[BB-HDMI-TDA998x-00A0.bb.org-overlays]
kernel:[5.4.106]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[dtb_overlay=/usr/lib/firmware/BOTIC-00A0.dtbo]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20210821.0-0~buster+20210821]
pkg:[bb-wl18xx-firmware]:[1.20210901.0-0~buster+20210901]
pkg:[kmod]:[26-1]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal input bluetooth netdev gpio admin spi iio i2c docker tisdk weston-launch xenomai cloud9ide pwm eqep remoteproc]
cmdline:[console=ttyO0,115200n8 snd_soc_botic.dai_format=16386 snd_soc_botic.blr_ratio=64 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
[    2.099272] remoteproc remoteproc0: wkup_m3 is available
[    2.472554] remoteproc remoteproc1: 4a334000.pru is available
[    2.474570] remoteproc remoteproc2: 4a338000.pru is available
[    2.479776] remoteproc remoteproc0: powering up wkup_m3
[    2.479936] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217148
[    2.483040] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
[    2.472554] remoteproc remoteproc1: 4a334000.pru is available
[    2.472778] pru-rproc 4a334000.pru: PRU rproc node /ocp/interconnect@4a000000/segment@0/target-module@300000/pruss@0/pru@34000 probed successfully
[    2.474570] remoteproc remoteproc2: 4a338000.pru is available
[    2.474762] pru-rproc 4a338000.pru: PRU rproc node /ocp/interconnect@4a000000/segment@0/target-module@300000/pruss@0/pru@38000 probed successfully
dmesg | grep pinctrl-single
[    1.709989] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    1.711905] gpio-of-helper ocp:cape-universal: ready
END
```

RobertCNelson, 2021-09-20:
Did you install the linux headers:

```
sudo apt install linux-headers-`uname -r`
```
Regards,

bernid, 2021-09-20:
Hi, 
Ofcourse.
I installed my new build:
linux-image-5.4.106_5.4.106-1_armhf.deb
linux-headers-5.4.106_5.4.106-1_armhf.deb
linux-libc-dev_5.4.106-1_armhf.deb

bernid, 2021-09-20:
> Did you build nativity or cross built the *.deb files?

This is my natively build of the newest RT kernel. 
`Linux beaglebone 5.4.106 #1 SMP PREEMPT_RT Sat Sep 18 22:16:51 CEST 2021 armv7l GNU/Linux`
I took about 20h on my BBB

bernid, 2021-09-21:
I've build kernel on my BBB many times before (last time 9 months ago) and had no trouble building the external module afterwards. For example:
```
git clone -b 5.4-rt --depth 1  https://github.com/beagleboard/linux
sudo make bindeb-pkg
```
It looks like some headers files are missing now.
Regards, 
Bern

orgua, 2022-04-04:
Was there any progress regarding this problem? I use the BBG and came from kernel 4.19 and wanted to update to 5.4 but my kernel module is failing with the same error as mentioned above. I used the kernel_update.sh from https://github.com/RobertCNelson/boot-scripts and installed headers afterwards, as well as "apt dist-upgrade". 

Current uname: linux-kernel-5.4.106-ti-r40

Kali-, 2022-09-19:
i'm facing this issue with 5.x kernel
i have take a look of many linux-headers-5.*_1buster_armhf.deb and all looks to be incomplete, Kconfig and many dirs are missing:

# ls -1 /usr/src/linux-headers-5.10.131-ti-r49
Makefile
Module.symvers
arch
include
scripts

4.19 kernel looks good

RobertCNelson, 2022-09-19:
this is a change in 5.10.x that messed all this up..

https://stackoverflow.com/questions/65972180/im-unable-to-build-the-simple-hello-world-kernel-module-from-kernel-5-10-11/65979628#65979628

https://www.kernel.org/doc/Documentation/kbuild/modules.txt

https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1906131

change KDIR..

Regards,

Kali-, 2022-09-20:
odd because the same issue is in your kernel 5.4.x 
maybe it is a different issue (I can open a new bug if u prefer), but definitely your 5.4.x and 5.10.x kernel headers are incomplete 

.#ls usr/src/linux-headers-5.4.106-ti-r41
Makefile       Module.symvers arch           include        scripts

.#ls usr/src/linux-headers-5.10.131-ti-r49
Makefile       Module.symvers arch           include        scripts

. # ls /usr/src/linux-headers-4.19.94-ti-r73/
Documentation  Makefile        arch   certs   drivers   fs       init  kernel  mm   samples  security  tools  virt
Kconfig        Module.symvers  block  crypto  firmware  include  ipc   lib     net  scripts  sound     usr

.#ls /usr/src/linux-headers-5.15.48-sunxi64
arch   certs   Documentation  fs       init  Kconfig  lib       mm              net      scripts   sound  usr
block  crypto  drivers        include  ipc   kernel   Makefile  Module.symvers  samples  security  tools  virt

Kind Regards,
Giulio