beagleboard/linux #269: Error while building uImage

CodeOn-ArK, 2022-02-25:
Hello
This is the error I get when compiling the Kernel
command used `  arm-make uImage dtbs modules LOADADDR=0x80008000 -j8` 
error : 
```
In file included from ./include/linux/bitmap.h:9,
                 from ./include/linux/nodemask.h:95,
                 from ./include/linux/mmzone.h:17,
                 from ./include/linux/gfp.h:6,
                 from ./include/linux/umh.h:4,
                 from ./include/linux/kmod.h:9,
                 from ./include/linux/module.h:13,
                 from crypto/asymmetric_keys/public_key.c:11:
In function ‘memcpy’,
    inlined from ‘pkey_pack_u32.part.0’ at crypto/asymmetric_keys/public_key.c:96:2:
./include/linux/string.h:397:25: error: call to ‘__write_overflow’ declared with attribute error: detected write beyond size of object passed as 1st parameter
  397 |                         __write_overflow();
```
I get this error several times like
```
In function ‘memset’,
    inlined from ‘btrfs_io_bio_init.part.0’ at fs/btrfs/extent_io.c:2859:2:
./include/linux/string.h:385:17: error: call to ‘__write_overflow’ declared with attribute error: detected write beyond size of object passed as 1st parameter
  385 |                 __write_overflow();
      |                 ^~~~~~~~~~~~~~~~~~
```
 ```
 CC      kernel/power/process.o
In file included from ./include/linux/bitmap.h:9,
                 from ./include/linux/cpumask.h:12,
                 from ./include/linux/rcupdate.h:31,
                 from ./include/linux/rculist.h:11,
                 from ./include/linux/pid.h:5,
                 from ./include/linux/sched.h:14,
                 from ./include/linux/cgroup.h:12,
                 from ./include/linux/memcontrol.h:13,
                 from mm/memcontrol.c:26:
In function ‘memset’,
    inlined from ‘uncharge_gather_clear.part.0’ at mm/memcontrol.c:6676:2:
./include/linux/string.h:385:17: error: call to ‘__write_overflow’ declared with attribute error: detected write beyond size of object passed as 1st parameter
  385 |                 __write_overflow();
      |                 ^~~~~~~~~~~
```~~~~~~~

RobertCNelson, 2022-02-25:
Hi @CodeOn-ArK sorry, we have not shipped "uImage" by default for many years now.. can you replicate this on mainline?  Sorry it's something i never test..

CodeOn-ArK, 2022-02-26:
Well Master gives this 😆 
```
❯ arm-make bb.org_defconfig uImage dtbs modules LOADADDR=0x80008000 -j8
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/confdata.o
  HOSTCC  scripts/kconfig/expr.o
  LEX     scripts/kconfig/lexer.lex.c
  YACC    scripts/kconfig/parser.tab.[ch]
  HOSTCC  scripts/kconfig/menu.o
  HOSTCC  scripts/kconfig/preprocess.o
  HOSTCC  scripts/kconfig/symbol.o
  HOSTCC  scripts/kconfig/util.o
  HOSTCC  scripts/kconfig/lexer.lex.o
  HOSTCC  scripts/kconfig/parser.tab.o
  HOSTLD  scripts/kconfig/conf
***
*** Can't find default configuration "arch/arm/configs/bb.org_defconfig"!
***
make[2]: *** [scripts/kconfig/Makefile:94: bb.org_defconfig] Error 1
make[1]: *** [Makefile:616: bb.org_defconfig] Error 2
make: *** [Makefile:350: __build_one_by_one] Error 2
```

RobertCNelson, 2022-02-26:
on master use "omap2plus_defconfig"

CodeOn-ArK, 2022-02-27:
This is the error now with master and omap2plus_defconfig
```
  HOSTCC  scripts/sorttable
In file included from /home/ark/linux_workspace/embedded_linux_workspace/ldd/downloads/gcc-linaro-12.0.0-2021.11-x86_64_arm-linux-gnueabihf/bin/../lib/gcc/arm-linux-gnueabihf/12.0.0/plugin/include/gcc-plugin.h:28,
                 from scripts/gcc-plugins/gcc-common.h:7,
                 from scripts/gcc-plugins/arm_ssp_per_task_plugin.c:3:
/home/ark/linux_workspace/embedded_linux_workspace/ldd/downloads/gcc-linaro-12.0.0-2021.11-x86_64_arm-linux-gnueabihf/bin/../lib/gcc/arm-linux-gnueabihf/12.0.0/plugin/include/system.h:702:10: fatal error: gmp.h: No such file or directory
  702 | #include <gmp.h>
      |          ^~~~~~~

```

RobertCNelson, 2022-02-27:
> /home/ark/linux_workspace/embedded_linux_workspace/ldd/downloads/gcc-linaro-12.0.0-2021.11-x86_64_arm-linux-gnueabihf/bin/../lib/gcc/arm-linux-gnueabihf/12.0.0/plugin/include/system.h:702:10: fatal error: gmp.h: No such file or directory
>   702 | #include <gmp.h>
>       |          ^~~~~~~

https://packages.debian.org/bullseye/libgmp-dev

```
sudo apt update ; sudo apt install libgmp-dev
```

CodeOn-ArK, 2022-03-01:
It compiles well, with even lesser number of warnings

RobertCNelson, 2022-03-01:
Cool, so no issues?? Closing?

CodeOn-ArK, 2022-03-01:
Is unable to boot the image is an issue of the tree, or I have to try harder?

RobertCNelson, 2022-03-01:
mainline should boot fine..

please post your full serial log.. (use j1)...

Regards,

CodeOn-ArK, 2022-03-02:
> Linux beaglebone 5.15.0-rc4-00015-g60a9483534ed #5 SMP  armv7l GNU/Linux

yay! boots well with `consoel=ttyO0` 
Thanks a lot!

movhdi, 2023-03-26:
Hi, I have downloaded the linux-5.10 branch, while using "bb.org_defconfig" I am receiving the same error compiling linux kernel.
I am using <arm-linux-gnueabihf-gcc> cross compile tool.

movhdi, 2023-03-26:
I installed libgmp-dev as you said, but it did not work. my log is:

> In file included from scripts/gcc-plugins/gcc-common.h:139,
>                  from scripts/gcc-plugins/arm_ssp_per_task_plugin.c:3:
> /home/movhdi/embedlinuxWS/Downloads/gcc-linaro-13.0.0-2022.11-x86_64_arm-linux-gnueabihf/bin/../lib/gcc/arm-linux-gnueabihf/13.0.0/plugin/include/builtins.h:23:10: fatal error: mpc.h: No such file or directory
>    23 | #include <mpc.h>
>       |          ^~~~~~~
> compilation terminated.
> make[2]: *** [scripts/gcc-plugins/Makefile:47: scripts/gcc-plugins/arm_ssp_per_task_plugin.so] Error 1

RobertCNelson, 2023-03-27:
@movhdi did you read your build error?

```
linux-gnueabihf/13.0.0/plugin/include/builtins.h:23:10: fatal error: mpc.h: No such file or directory
23 | #include <mpc.h>
| ^~~~~~~
compilation terminated.
```

mpc.h not gmp.h....

That's another library.. https://packages.debian.org/bullseye/libmpc-dev

```
sudo apt update ; sudo apt install libmpc-dev
```

movhdi, 2023-03-27:
Thank you, that is right. I installed this lib but the problem persisted, I also changed the toolchain from version 13 to 7.5 of linaro, and oit compiled successfully.