beagleboard/linux #276: 5.18.0.rc4+ doesn't boot on Beaglebone black rev.C

plcs2eco, 2022-11-28:
Hi,
I am trying to boot from microSD on beaglebone black.
- microSD partition 1 fat16, u-boot.img & boot.scr & MLO & u-boot-spl.bin
- microSD partition 2 ext3, rootfile system
- TFTP server (uBuntu) , uImage, am335x_boneblack.dtb

I downloaded zip file from this site around Nov 5th, 2022.   Using below combination
- Linux-5.18.0-rc4+  (I compiled with bb.org_defconfig)
     make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- uImage dtbs LOADADDR=0x80008000
- busybox-1.35.0
- gcc-linaro-7.5.0-2019.12-x86_64_arm-linux-gnueabihf
- u-boot-2022.10 (I compiled with am335x_evm_defconfig)

I install all generated modules into the rootfile system which was generated by busybox.  And I added /dev, /etc, /root, /sys, /config and /proc.   Kernel started but became kernel panic and stopped and didn't even reach the point where it requires those directories with below errors. 
[    9.463643] /dev/root: Can't open blockdev                                   
[    9.475615] VFS: Cannot open root device "mmcblk0p2" or unknown-block(0,0): 6

mmcblk0p2 is the second partition on the microSD where the rootfile system resides.
 
I searched the solution but couldn't figure it out.   Could anyone help me?   Am I using wrong kernel for beaglebone black rev C?

[boot_scr_from_sd.txt](https://github.com/beagleboard/linux/files/10101035/boot_scr_from_sd.txt)

RobertCNelson, 2022-11-28:
```
setenv bootargs console=ttyO0,115200n8 noinitrd rootfstype=ext3 root=/dev/mmcblk0p2 rw mem=512M boot_delay=3 rootdelay=5
```

Why the odd boot_delay and rootdelay values?

just swap them for `rootwait`..

`console=ttyO0,115200n8`  you should use  ``console=ttyS0,115200n8`

Please post your full serial boot log with those changes..

PS... why Linux-5.18.0-rc4+, smells like a random choice..  If you need 5.18.x use 5.18.19 over 5.18.0-rc4...

Regards,

plcs2eco, 2022-11-29:
Hi Nelson,
Thank you very much for your advice.   I have been battling this issue for over a week by myself by googling and a lot of trial&errors.  

1) I replaces "boot_delay " and "root_delay" with "rootwait". The error message changed to the below message.  And the message doesn't change even after  minutes.

[   33.760875] vmmcsd_fixed: disabling

And D2 LED started flashing.  So it might have been successfully booted but I can't see in minicom.  This phenomena is same with ttyO0 or ttyS0. 

2) Linux-5.18.0-rc4+
I don't know why I am using this Linux-5.18.0-rc4+.  I was going through Udemy course.  And this link was recommended to download linux kernel source files for BBB.   https://github.com/beagleboard/linux
I downloaded as zip and extracted and built the kernel.  That happened to be Linux-5.18.0-rc4+.  It was not my choice.

3) ttyO0
In the Udemy course, they use ttyO0 instead of ttyS0.  I kind of knew that ttyO0 was replaced by ttyS0 in newer system.  But in the serial boot log message, kernel replaced ttyO0 witth ttyS0.  So I thought that it doesn't hurt keeping ttyO0.  The below log is with
[boot_scr_from_sd.txt](https://github.com/beagleboard/linux/files/10109794/boot_scr_from_sd.txt)
 ttyS0.  

<1> Does D2 LED flashing mean that it successfully booted?  I saw the schematic and D2 is connected to USR0 but couldn't understand why D2 flashes.  I also checked BBB system ref document.  No clue found.

<2> How can I shorten the boot time instead of Why does it take long 33 seconds?

<3> From Udemy course, once it booted, I am supposed to see # console on minicom and to be able to use bin/sbin command.   What am I doing wrong?

<4> This is BBB without wifi.  Why is it trying to load the below file?
[    3.562473] cfg80211: failed to load regulatory.db

<4> cpsw or eth0 is not recognized in the log.   So I can't use NFS mounting rootfs instead of microSD.  I checked menuconfig(.config) and it appears to me that net is selected as shown below.   How can I fix it?

 Symbol: PHY_TI_GMII_SEL [=y]                                                                                                                                                            
  Type  : tristate                                                                                                                                                                         
  Defined at drivers/phy/ti/Kconfig:107                                                                                                                                                     
  Selects: GENERIC_PHY [=y] && REGMAP [=y]                                                                                                                                                    
  Implied by [y]:                                                                                                                                                                              
  - TI_CPSW [=y] && NETDEVICES [=y] && ETHERNET [=y] && NET_VENDOR_TI [=y] && (ARCH_DAVINCI [=n] || ARCH_OMAP2PLUS [=y] || COMPILE_TEST [=n]) && (TI_CPTS [=y] || !TI_CPTS [=y])  

I am a beginner of using github.  so I am sorry if I asked many questions or asked wrong question to ask in here.   If so, please let me know and I will do as told.

Your help would be truly appreciated.
Best regards,
Junji

plcs2eco, 2022-11-29:
Here is the full serial boot log.
--------------------------------------------------------------------------------
U-Boot SPL 2022.10 (Nov 25 2022 - 14:10:15 +0900)                                                                                   
Trying to boot from MMC1                                                                                                            
                                                                                                                                    
                                                                                                                                    
U-Boot 2022.10 (Nov 25 2022 - 14:10:15 +0900)                                                                                       
                                                                                                                                    
CPU  : AM335X-GP rev 2.1                                                                                                            
Model: TI AM335x BeagleBone Black                                                                                                   
DRAM:  512 MiB                                                                                                                      
Core:  160 devices, 18 uclasses, devicetree: separate                                                                               
WDT:   Started wdt@44e35000 with servicing (60s timeout)                                                                            
NAND:  0 MiB                                                                                                                        
MMC:   OMAP SD/MMC: 0, OMAP SD/MMC: 1                                                                                               
Loading Environment from FAT... Unable to read "uboot.env" from mmc0:1...                                                           
<ethaddr> not set. Validating first E-fuse MAC                                                                                      
Net:   eth2: ethernet@4a100000, eth3: usb_ether                                                                                     
Hit any key to stop autoboot:  0                                                                                                    
switch to partitions #0, OK                                                                                                         
mmc0 is current device                                                                                                              
Scanning mmc 0:1...                                                                                                                 
Found U-Boot script /boot.scr                                                                                                       
621 bytes read in 4 ms (151.4 KiB/s)                                                                                                
## Executing script at 80000000                                                                                                     
**************Wowowo ichiri************                                                                                             
Booting from microSD ...                                                                                                            
link up on port 0, speed 100, full duplex                                                                                           
Using ethernet@4a100000 device                                                                                                      
TFTP from server 192.168.27.1; our IP address is 192.168.27.2                                                                       
Filename 'uImage'.                                                                                                                  
Load address: 0x82000000                                                                                                            
Loading: #################################################################                                                          
         #################################################################                                                          
         #################################################################                                                          
         #################################################################                                                          
         #################################################################                                                          
         #################################################################                                                          
         #################################################################                                                          
         #################################################################                                                          
         #################################################################                                                          
         #################################################################                                                          
[... cut ...]
[    3.540216] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'                                                           
[    3.547322] platform regulatory.0: Direct firmware load for regulatory.db failed with error -2                                   
[    3.557217] Waiting for root device /dev/mmcblk0p2...                                                                            
[    3.562473] cfg80211: failed to load regulatory.db                                                                               
[   33.760896] vmmcsd_fixed: disabling

RobertCNelson, 2022-11-29:
Thanks for the printdump..

Well everything looks fine except for the mmc not coming up..  So it's either 5.18.0-rc4 or your defconfig..

Ah... Udemy, that makes sense.. Why don't you ask them?

Anyways throw everything you've done out, your just going to spin your wheels.

```
git clone -b 5.10 https://github.com/beagleboard/linux
```

^ that'll get you a proper linux source to build from..

Stop downloading *.zip files from github..

plcs2eco, 2022-11-30:
Hi Robert,
It worked!     
Here is what I did.
1) did git clone to get 5.10
2) built kernel image as 
sudo make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- uImage dtbs LOADADDR=0x80008000 -j4
3) built modules
sudo make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- modules -j4

4) built busybox as build-static in menuconfig
5)  installed modules from linux to root file system
6) added rcS for init
 
And it showed the # console.   I am super excited now.  Yesterday, I downloaded zip and extracted and built the kernel image and modules and installed modules into root file system made by busybox.  I didn't choose "build static".   This improved from the last error and didn't mount fs last time.  It mounted and free unnecessary memory of kernel.  But it failed on init process from init/main.c.  I even tried to include init=/etc/init.d/rcS in cmdline.  I read https://docs.kernel.org/admin-guide/init.html.  I think that sbin/init was not properly built.  I thought that only difference in "download zip file" or "git clone" is same except for git clone install git access.  So I am suspecting that building busybox with static made it work from the init not found problem.

I need a lot of customization in the future so I need busybox.  When I ran jenkins_build.sh, I realized that it builds everything without any room to work with busybox.  Or at least I couldn't figure out how.  So I did this way.  Anyway, I am very happy and truly appreciate your professional help.  Just in case I attach my log.   I added debug to bootargs in order to get little bit more information.

By the way, I posted to get help few times at Udemy including this problem 4 days ago but anyone unfortunately never replied.  I was googling and reading a lot of sites and became desperate and posted here knowing my question was too dummy. 

Anyway, Thank you very much, Robert!
Junji

plcs2eco, 2022-11-30:
U-Boot SPL 2022.10 (Nov 25 2022 - 14:10:15 +0900)
Trying to boot from MMC1

U-Boot 2022.10 (Nov 25 2022 - 14:10:15 +0900)

CPU  : AM335X-GP rev 2.1
Model: TI AM335x BeagleBone Black
DRAM:  512 MiB
Core:  160 devices, 18 uclasses, devicetree: separate
WDT:   Started wdt@44e35000 with servicing (60s timeout)
NAND:  0 MiB
MMC:   OMAP SD/MMC: 0, OMAP SD/MMC: 1
Loading Environment from FAT... Unable to read "uboot.env" from mmc0:1... 
<ethaddr> not set. Validating first E-fuse MAC
Net:   eth2: ethernet@4a100000, eth3: usb_ether
Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr
641 bytes read in 4 ms (156.3 KiB/s)
## Executing script at 80000000
**************Wowowo ichiri************
Booting from microSD ...
link up on port 0, speed 100, full duplex
Using ethernet@4a100000 device
TFTP from server 192.168.27.1; our IP address is 192.168.27.2
Filename 'uImage'.
Load address: 0x82000000
Loading: #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         ##############################################  0 Bytes
         1.4 MiB/s
done
Bytes transferred = 10547776 (a0f240 hex)
link up on port 0, speed 100, full duplex
Using ethernet@4a100000 device
TFTP from server 192.168.27.1; our IP address is 192.168.27.2
Filename 'am335x-boneblack.dtb'.
Load address: 0x88000000
Loading: ###################  0 Bytes
[... 319 lines cut ...]
Jan  1 00:00:01 (none) user.info kernel: [    0.000000] rcu: AdjustJan  1 00:00:01 (none) user.info kernel: [    0.049705] MountpoiLoading kernel modules : Jan  1 00:00:01 (none) user.info kerns
an  1 00:00:01 (none) user.warn kernel: [    4.995527] at24 0-0J[    6.079275] using random self ethernet address
an  1 00:00:01 (none) user.info kernel: [    5.328009] tilcdc 4J[    6.085373] using random host ethernet address
an  1 00:00:01 (none) user.info kernel: [    5.492622] Not actiJ[    6.097336] usb0: HOST MAC b2:a7:1b:5e:b4:a4
an  1 00:00:02 (none) user.err kernel: [    6.067841] Module has[    6.105343] usb0: MAC c2:ed:a8:18:28:c5
 invalid ELF structures
Jan  1 00:00:02 (none) user.warn kernel[    6.115504] using random self ethernet address
: [    6.079275] using random self ethernet address
Jan  1 00:0[    6.125295] using random host ethernet address
0:02 (none) user.warn kernel: [    6.085373] using random host e[    6.135450] g_ether gadget: Ethernet Gadget, version: Memorial Day 2008
thernet address
Jan  1 00:00:02 (none) user.info kernel: [    6[    6.147667] g_ether gadget: g_ether ready
.097336] usb0: HOST MAC b2:a7:1b:5e:b4:a4
Jan  1 00:00:02 (none) user.info kernel: [    6.105343] usb0: MAC c2:ed:a8:18:28:c5
Jan  1 00:00:02 (none) user.warn kernel: [    6.115504] using random self ethernet address
Jan  1 00:00:02 (none) user.warn kernel: [    6.125295] using random host ethernet address
Jan  1 00:00:02 (none) user.info kernel: [    6.135450] g_ether gadget: Ethernet Gadget, version: Memorial Day 2008
Jan  1 00:00:02 (none) user.info kernel: [    6.147667] g_ether gadget: g_ether ready
OK
Starting network: ifup: can't open '/etc/network/interfaces': No such file or directory
FAIL

Please press Enter to activate this console. 
Jan  1 00:00:13 (none) daemon.info : starting pid 154, tty '': '-/bin/sh'
/ #

closed 2022-11-30 by plcs2eco