beagleboard/linux #283: BeaglePlay - Poky (kirkstone) - eth0 does not come up - mdio error -22 during boot-up

sckulkarni246, 2023-08-15:
Some info as below.

- Yocto version - `kirkstone`
- Bitbake layer used - `meta-ti` - latest commit used
- Distro - `poky`
- Machine - `beagleplay` (device-tree is `k3-am625-beagleplay.dts` - changes done)
- Image - `core-image-base`
- Changes done to kernel or device-tree - NO
- Expectation - `eth0` interface should come up, ethernet PHY should be discovered correctly
- Observation - mdio errors are observed during boot-up. Log as below.

```
[    3.145005] 2850000.serial: ttyS3 at MMIO 0x2850000 (irq = 283, base_baud = 3000000) is a 8250
[    3.155275] 2860000.serial: ttyS4 at MMIO 0x2860000 (irq = 284, base_baud = 3000000) is a 8250
[    3.166140] davinci_mdio 8000000.ethernet:mdio@f00: Configuring MDIO in manual mode
[    3.173866] davinci_mdio: probe of 8000000.ethernet:mdio@f00 failed with error -22
[    3.181651] am65-cpsw-nuss 8000000.ethernet: initializing am65 cpsw nuss version 0x6BA01103, cpsw versi
on 0x6BA81103 Ports: 3 quirks:00000006
[    3.194488] am65-cpsw-nuss 8000000.ethernet: initialized cpsw ale version 1.5
[    3.201629] am65-cpsw-nuss 8000000.ethernet: ALE Table size 512
[    3.208529] am65-cpsw-nuss 8000000.ethernet: CPTS ver 0x4e8a010c, freq:0, add_val:7 pps:0
[    3.219177] am65-cpsw-nuss 8000000.ethernet: set new flow-id-base 19
```

The pre-loaded debian image works as expected - no mdio errors are seen and `eth0` interface comes up correctly.
```
[    6.581937] am65-cpsw-nuss 8000000.ethernet: down msc_sl e0000000 tmo 0
[    6.587901] am65-cpsw-nuss 8000000.ethernet: set new flow-id-base 19
[    6.593028] am65-cpsw-nuss 8000000.ethernet eth0: PHY [gpio-0:00] driver [Generic PHY] (irq=POLL)
[    6.593062] am65-cpsw-nuss 8000000.ethernet eth0: configuring for phy/rgmii-rxid link mode
``` 

Debug efforts so far are as below.
- Made changes to the mdio node such that it is similar to that of the AM625-SK from TI - no change in behavior observed.
- The uname output on the debian image is as below.
```
debian@BeaglePlay:~$ uname -r
5.10.153-ti-arm64-r86
```
However, I can't find a branch with this name so I am unable to compare the device-trees and the kernel configurations as such.

Look forward to pointers to understand and root-cause this behavior.

Regards,
Shashank

closed 2024-08-16 by sckulkarni246

spb06, 2024-08-22:
Hi there, 

We are going through something pretty similar to this issue and are wondering if this was ever resolved. Thanks!

sckulkarni246, 2024-12-12:
Seeing this pretty late - a kernel upgrade helped. The Ethernet PHY on the BeaglePlay is a bit iffy - needs more time to get auto-detected. The patch from TI resolves this.
https://lore.kernel.org/u-boot/20240328150930.GB3442575@bill-the-cat/T/#m93d91ad01035278ad8768a33000988a6a3e1b570