beagleboard/linux #284: Where is the documentation to put the new kernel on the beaglebone black

graftonskaggs, 2023-11-29:
I managed to build a kernel using the steps in the wiki. I then copied the kernel, updated the uEnv.txt and put the am335x-bone.dts in the dts folder but when I rebooted it started complaining about some other device trees for uboot and it never loads. 

What am I missing? Is there any step by step instructions for putting the kernel on the Black

 U-Boot 2019.04-00002-g31a8ae0206 (May 13 2020 - 09:26:17 -0500), Build: jenkins-github_Bootloader-Builder-139

CPU  : AM335X-GP rev 2.1
I2C:   ready
DRAM:  512 MiB
No match for driver 'omap_hsmmc'
No match for driver 'omap_hsmmc'
Some drivers were not found
Reset Source: Power-on reset has occurred.
RTC 32KCLK Source: External.
MMC:   OMAP SD/MMC: 0, OMAP SD/MMC: 1
Loading Environment from EXT4... Card did not respond to voltage select!
Board: BeagleBone Black
<ethaddr> not set. Validating first E-fuse MAC
BeagleBone Black:
BeagleBone: cape eeprom: i2c_probe: 0x54:
BeagleBone: cape eeprom: i2c_probe: 0x55:
BeagleBone: cape eeprom: i2c_probe: 0x56:
BeagleBone: cape eeprom: i2c_probe: 0x57:
Net:   eth0: MII MODE
cpsw, usb_ether
Press SPACE to abort autoboot in 0 seconds
board_name=[A335BNLT] ...
board_rev=[00C0] ...
Card did not respond to voltage select!
Card did not respond to voltage select!
Card did not respond to voltage select!
gpio: pin 56 (gpio 56) value is 0
gpio: pin 55 (gpio 55) value is 0
gpio: pin 54 (gpio 54) value is 0
gpio: pin 53 (gpio 53) value is 1
Card did not respond to voltage select!
Card did not respond to voltage select!
switch to partitions #0, OK
mmc1(part 0) is current device
Scanning mmc 1:1...
gpio: pin 56 (gpio 56) value is 0
gpio: pin 55 (gpio 55) value is 0
gpio: pin 54 (gpio 54) value is 0
gpio: pin 53 (gpio 53) value is 1
switch to partitions #0, OK
mmc1(part 0) is current device
gpio: pin 54 (gpio 54) value is 1
Checking for: /uEnv.txt ...
Checking for: /boot.scr ...
Checking for: /boot/boot.scr ...
Checking for: /boot/uEnv.txt ...
gpio: pin 55 (gpio 55) value is 1
1997 bytes read in 15 ms (129.9 KiB/s)
Loaded environment from /boot/uEnv.txt
Checking if uname_r is set in /boot/uEnv.txt...
gpio: pin 56 (gpio 56) value is 1
Running uname_boot ...
loading /boot/vmlinuz-3.8.13 ...
5585760 bytes read in 364 ms (14.6 MiB/s)
debug: [enable_uboot_overlays=1] ...
debug: [enable_uboot_cape_universal=1] ...
debug: [uboot_base_dtb_univ=am335x-boneblack-uboot-univ.dtb] ...
uboot_overlays: [uboot_base_dtb=am335x-boneblack-uboot-univ.dtb] ...
debug: unable to find [am335x-boneblack-uboot-univ.dtb] using [am335x-boneblack-uboot.dtb] instead ...
debug: [uboot_base_dtb_univ=am335x-boneblack-uboot.dtb] ...
uboot_overlays: [uboot_base_dtb=am335x-boneblack-uboot.dtb] ...
loading /boot/dtbs/am335x-boneblack.dtb ...
26118 bytes read in 17 ms (1.5 MiB/s)
uboot_overlays: [fdt_buffer=0x60000] ...
uboot_overlays: loading /lib/firmware/BB-ADC-00A0.dtbo ...
867 bytes read in 33 ms (25.4 KiB/s)
uboot_overlays: loading /lib/firmware/BB-BONE-eMMC1-01-00A0.dtbo ...
1584 bytes read in 140 ms (10.7 KiB/s)
uboot_overlays: loading /lib/firmware/BB-HDMI-TDA998x-00A0.dtbo ...
4915 bytes read in 44 ms (108.4 KiB/s)
uboot_overlays: loading /lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo ...
3801 bytes read in 232 ms (15.6 KiB/s)
failed on fdt_overlay_apply(): FDT_ERR_NOTFOUND
debug: [console=ttyO0,115200n8 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet] ...
debug: [bootz 0x82000000 - 88000000] ...
ERROR: Did not find a cmdline Flattened Device Tree
Could not find a valid device tree
** Invalid partition 2 **
** Invalid partition 3 **
** Invalid partition 4 **
** Invalid partition 5 **
** Invalid partition 6 **
** Invalid partition 7 **
starting USB...
USB0:   scanning bus 0 for devices... 1 USB Device(s) found
       scanning usb for storage devices... 1 Storage Device(s) found
cpsw Waiting for PHY auto negotiation to complete......... TIMEOUT !
using musb-hdrc, OUT ep1out IN ep1in STATUS ep2in
MAC e0:ff:f1:af:1e:09
HOST MAC de:ad:be:af:00:00
RNDIS ready
musb-hdrc: peripheral reset irq lost!
high speed config #2: 2 mA, Ethernet Gadget, using RNDIS
The remote end did not respond in time.missing environment variable: pxeuuid
Retrieving file: pxelinux.cfg/01-e0-ff-f1-af-1e-09
cpsw Waiting for PHY auto negotiation to complete......... TIMEOUT !
using musb-hdrc, OUT ep1out IN ep1in STATUS ep2in
MAC e0:ff:f1:af:1e:09
HOST MAC de:ad:be:af:00:00
RNDIS ready
high speed config #2: 2 mA, Ethernet Gadget, using RNDIS
The remote end did not respond in time.Retrieving file: pxelinux.cfg/C0A80102
cpsw Waiting for PHY auto negotiation to complete......... TIMEOUT !
using musb-hdrc, OUT ep1out IN ep1in STATUS ep2in
MAC e0:ff:f1:af:1e:09
HOST MAC de:ad:be:af:00:00
RNDIS ready
high speed config #2: 2 mA, Ethernet Gadget, using RNDIS
The remote end did not respond in time.Retrieving file: pxelinux.cfg/C0A8010
cpsw Waiting for PHY auto negotiation to complete......... TIMEOUT !
using musb-hdrc, OUT ep1out IN ep1in STATUS ep2in
MAC e0:ff:f1:af:1e:09
HOST MAC de:ad:be:af:00:00
RNDIS ready
high speed config #2: 2 mA, Ethernet Gadget, using RNDIS

RobertCNelson, 2023-11-29:
```
loading /boot/vmlinuz-3.8.13 ...
5585760 bytes read in 364 ms (14.6 MiB/s)
debug: [enable_uboot_overlays=1] ...
debug: [enable_uboot_cape_universal=1] ...
```
For old 3.8.13, comment out these in /boot/uEnv.txt

```
enable_uboot_overlays=1
enable_uboot_cape_universal=1
```

```
#enable_uboot_overlays=1
#enable_uboot_cape_universal=1
```

DeepakKumarShah, 2024-06-16:
https://labs.dese.iisc.ac.in/embeddedlab/building-the-linux-kernel-cross-compiling-for-a-beaglebone-black-2/