beagleboard/linux #286: Enabling CONFIG_USB_TI_CPPI41_DMA and disabling CONFIG_MUSB_PIO_ONLY.

davenmccalla-diagraph, 2024-02-09:
This question is related to issue #131 which has been closed.

I'm occasionally seeing sluggish behavior due to a musb irq:
[image]

If I enable CONFIG_USB_TI_CPPI41_DMA and disable CONFIG_MUSB_PIO_ONLY as described in issue #131, how can I be sure that I won't be affected by the "instability" or "very buggy" behavior described in the following links?

[https://www.phytec.eu/fileadmin/legacy/downloads/Manuals/TN-041e_1.pdf](url)
[https://forum.beagleboard.org/t/rndis-usb-tethering-performance-issue/36290](url)

Would a day or two of testing without seeing any problems be enough?

RobertCNelson, 2024-02-09:
'problems' are usb 'device' specific..  Normally we'd see issues with wifi usb devices plugged into the am335x..

So if your plan is to only plug into devices and use the usb gadget connection.. should be fine..

if you plan to use any random wifi device, it'll fail..

Regards,

davenmccalla-diagraph, 2024-02-09:
That's what I was afraid you'd say. The [RTW_CMD_THREAD] just below [irq/195-musb-hd] in the screen shot above is a thread for rtl8723bu wifi dongle. That was with CONFIG_MUSB_PIO_ONLY enabled.

I've been running with CONFIG_USB_TI_CPPI41_DMA enabled for a day with no issues. **If there is a potential issue, would I have seen it already? Or could it take days or weeks to show up?**

I have seen these messages show up when I insert a USB flash drive with DMA enabled. Not sure if it's just a coincidence. The flash drive appears to be working.
[83639.724250] F2FS-fs (sda): Can't find valid F2FS filesystem in 1th superblock
[83639.739171] F2FS-fs (sda): Can't find valid F2FS filesystem in 2th superblock
[83639.752825] F2FS-fs (sda): Can't find valid F2FS filesystem in 1th superblock
[83639.762741] F2FS-fs (sda): Can't find valid F2FS filesystem in 2th superblock

closed 2024-02-09 by davenmccalla-diagraph